OPCFoundation/UA-.NETStandard #968: OPC UA Data Access Client problem

DmitryFarberov, 2020-05-06:
I connect to a server and create a monitored item. I then disconnect and repeat the steps. When I attempt to monitor the exact same item I get this:

**_EXCEPTION (ObjectDisposedException)
--
Cannot access a disposed object.Object name: 'TransportChannel is not  available.'.
STACK TRACE
at Opc.Ua.ClientBase.get_UseTransportChannel()at  Opc.Ua.SessionClient.CreateMonitoredItems(RequestHeader requestHeader, UInt32  subscriptionId, TimestampsToReturn timestampsToReturn,  MonitoredItemCreateRequestCollection itemsToCreate,  MonitoredItemCreateResultCollection& results, DiagnosticInfoCollection&  diagnosticInfos)at Opc.Ua.Client.Subscription.CreateItems()at  Opc.Ua.Client.Subscription.ApplyChanges()at  Quickstarts.DataAccessClient.MainForm.Browse_MonitorMI_Click(Object sender,  EventArgs e) in D:\Sourcetree\OPC  UA\UA-.NETStandard\SampleApplications\Workshop\DataAccess\Client\MainForm.cs:line  586_**

If I exit the client and try again, it will work the first time around.

koepalex, 2020-05-09:
I've prepared a simply fix, that prevent the crash, please see linked PR.

AlinMoldovean, 2020-05-11:
Fixed in #972

closed 2020-05-11 by AlinMoldovean