OPCFoundation/UA-.NETStandard #971: DLLs are could not found errors

yagizhanNY, 2020-05-08:
Hello everyone,

I have some problem about building a solution. When I build UA Reference solution I have some errors like screenshot at bellow. 

[image]

I tried to unload/reload projects and change framework targets to .net framework 4.8 but it isn't work. What should I do for these issues? I did some search in internet about this but I can't find any solution for this.

Have a healthy days.

koepalex, 2020-05-08:
Look in the build output why the Opc.Ua.Core could not be build, without this assembly the other could not be generated. 

Did you restore the nuget packages (at solution level) before building?

yagizhanNY, 2020-05-08:
Hi @koepalex  I changed the framework target for all libraries to 4.6.2 and it is working fine now. Thanks for the answer.

closed 2020-05-08 by yagizhanNY