OPCFoundation/UA-.NETStandard #973: A little document would be helpfull

torsilver, 2020-05-11:
I am trying to connect a Siemens OPC UA server.
And I got some strange nambers like below. Then I can read and write values.

AttributeId = 13
ns = 3

Where can I get some documents about those numbers?

koepalex, 2020-05-11:
This is defined by [OPC Specfication Part 3 - AddressModel](https://reference.opcfoundation.org/v104/Core/docs/Part3/)

Each server handles different namespaces accessed by index, you need to read the **NamespaceArray** from server node.

Each node is classified by namespace id and attribute id. To get a readable name you have to read **DisplayName** attribute.

torsilver, 2020-05-11:
Ok thanks

On 05/11/2020 17:40, Alexander Köpke wrote:

This is defined by OPC Specfication Part 3 - AddressModel

Each server handles different namespaces accessed by index, you need to read the NamespaceArray from server node.

Each node is classified by namespace id and attribute id. To get a readable name you have to read DisplayName attribute.

closed 2020-05-11 by torsilver