OPCFoundation/UA-.NETStandard #976: how do I prohibit anonymous users from accessing certain nodes?

fsodano1, 2020-05-13:
Hello everyone!
I'm developing a server and I have to regulate access to nodes, according to this policy:
- authenticated users: complete access to every node in the address space;
- anonymous users: read access on a restricted set of nodes;
How can i manage this situation?

AlinMoldovean, 2020-05-14:
Hi @fsodano1 ,

For restricting the Read/Write access of some variable nodes depending on the user you can have a look at _UserAuthenticationServer_ sample where anonymous user are restricted to write a specific variable node.

For using a role based access system, the upcoming PR #977 will provide support for using RolePermissions and AccessRestrictions attributes in a server.

fsodano1, 2020-05-15:
Hi @AlinMoldovean ,
I saw UserAuthenticationServer sample but it only reject all writes if no user is provided, if i understand correcty.
The upcoming PR#977 lends itself well to my purpose.  
When you plan to release on nuget a new pkg with this feature?
Thanks for the help.

AlinMoldovean, 2020-05-20:
@fsodano1 ,

A NuGet release containing this feature will come within a few days.

closed 2020-05-20 by AlinMoldovean