OPCFoundation/UA-.NETStandard #987: Ref Client should accept no cert encoded as NULL or empty bytestring

mregen, 2020-05-19:
Validate Mantis issue 5593:

There’s written in the OPC UA standard (https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.2/) that “If the securityPolicyUri is NONE and none of the UserTokenPolicies requires encryption, the Client shall ignore the ApplicationInstanceCertificate.” But some clients, including OPC Foundation UA Reference Sample client 1.4.360.33, do not ignore the certificate value and requires NULL byteString (FF FF FF FF) instead of empty byteString (00 00 00 00).

AlinMoldovean, 2020-05-20:
Can we have access to the server with the described behavior so we can test this?

AlinMoldovean, 2020-05-27:
I did not had access to the server with the specified behavior, but I changed our server to reproduce the condition.
The client skips the certificate validation for both Null and empty byteString values in ServerCertificate.

https://github.com/OPCFoundation/UA-.NETStandard/blob/be65403945d4f0bf366f3299222fcb6cb08d6cb5/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2283

closed 2020-05-27 by AlinMoldovean