OPCFoundation/UA-.NETStandard #991: Sample: Basic Event

PureIso, 2020-05-24:
I am currently learning OPC-UA .NetStandard. I can't seem to understand how to write a basic event.
I am able to monitor variables but I would also like to monitor when a method has been called.
The main issue I am having is knowing how to implement the server side event.

Would it be possible to add a basic event to the sample as the current example is difficult to understand.

AlinMoldovean, 2020-05-29:
Hi @PureIso ,

In order to notify clients when a specific method has been called you need to report an event after processing the method call handler at application level.

A code sample:
```
        private ServiceResult OnStopMonitoring(ISystemContext context, MethodState method, IList<object> inputArguments, IList<object> outputArguments)
        {
            // TODO: Process the method call
	    // 
			
	    // Create an event
            BaseEventState stopMonitoringEvent = new BaseEventState(method);

            stopMonitoringEvent.Initialize(SystemContext, method, EventSeverity.Medium,
                new LocalizedText("en", "Underlying system monitoring was stopped."));

            // Report the event at Server level
            Server.ReportEvent(stopMonitoringEvent); 

            return ServiceResult.Good;
        }
```

PureIso, 2020-05-29:
@AlinMoldovean 
Thank you very much for the reply.
I have the following as client:
```cs
MonitoredItem monitoredItem = new MonitoredItem
                {
                    StartNodeId = nodeId,
                    AttributeId = Attributes.EventNotifier,
                    MonitoringMode = MonitoringMode.Reporting,
                    SamplingInterval = 0,
                    QueueSize = 1000,
                    DiscardOldest = true,
                    Filter = new EventFilter()
                };
                monitoredItem.Notification += callback;
```
The problem now is that the callback never triggers?

AlinMoldovean, 2020-06-03:
@PureIso ,

You need to create the event monitored item on the _Server_ node and set a correct _EventFilter_

closed 2020-06-12 by PureIso