OPCFoundation/UA-.NETStandard #994: Should self-signed certificates be End Entity certificates?

MD-V, 2020-05-27:
All certificates created by the ApplicationInstance https://github.com/OPCFoundation/UA-.NETStandard/blob/300d9ffd62d117e06988687ced6f7c6547d39b23/SampleApplications/SDK/Opc.Ua.Configuration/ApplicationInstance.cs#L1277 are not assigned to any CA. 
I think this leads to BadCertificateChainIncomplete erros in the UAExpert. Which shouldn't occure because self-signed certs have no parent CA.

As stated on [stackexchange](https://security.stackexchange.com/a/82890) certificates which are not assigned to a CA should be End Entity certificates.

BouncyCastle can do this via: [setIsEndEntity](https://www.bouncycastle.org/docs/pkixdocs1.4/org/bouncycastle/cert/path/CertPathValidationContext.html#setIsEndEntity(boolean))

Are there any reasons why the self-signed certificates aren't end entity certificates?

mregen, 2020-05-31:
Hi @MD-V, I think your point is correct, the self signed certs should be tagged as CA, not end entity.
The RFC states only CA certs should have the KeyCertSign set, but here we set the flag and its an end entity.
From IOP standpoint I see both self signed certs as end entity and as CA certs. And I believe UA Expert is happy once the KeyCertSign flag is set. Has this changed recently?
So I think we should change the default to set the CA flag for the self signed certs.

mregen, 2020-06-08:
PR can be added after repo split: https://github.com/mregen/UA-.NetStandardLibrary/tree/selfsignedcerts

closed 2020-10-11 by mregen