OPCFoundation/UA-.NETStandard #995: GetIssuers() in CertificateValidator.cs returns false inappropriately

DmitryFarberov, 2020-05-27:
The GetIssuers() method starting on line 507 returns false even if the issuer is set to a valid reference on line 525:

                issuer = await GetIssuer(certificate, m_issuerCertificateList, m_issuerCertificateStore, true);

I got around the problem by adding an if clause on line 531:

                else
                 {
                      isTrusted = true;
                  }

mregen, 2020-05-31:
Hi @DmitryFarberov, thats exactly the point, if a certificate is in the issuer list the chain it is *not* trusted. The issuer cert is only used to validated the chain as a whole.
One cert in the chain must be in the trusted list to be trusted.

AlinMoldovean, 2020-06-04:
Hi @DmitryFarberov,

A specific compliance requirement of the OPC UA protocol allows a certificate to be considered trusted only if at least one of the CAs from the certificate chain is in the "trusted peer" certificates store.

As @mregen mentioned, the "Issuer" certificates store is only used for allowing the complete CA chain to be validated.
The name _**Trusted**IssuerCertificates_ is kept for backward compatibly (because in the initial version was using this store for trusted CAs) .

In the actual version you need place a CA certificate in the _TrustedPeerCertificates_ store in order to trust its issued certificates, like you did at point 2.
So the logic of _GetIssuers()_ method is correct.

Thanks!

DmitryFarberov, 2020-06-04:
@AlinMoldovean,

Thank you for the explanation. I will close the issue.

closed 2020-06-04 by DmitryFarberov