OPCFoundation/UA-.NETStandard #999: OPC Ua client in C#

flamurbreznica93, 2020-05-28:
Hi ! 

I use visual studio and I am trying to make a opc ua client. It works, gets connected and so on. But it is very slow on detecting that it needs to be reconnected. I followed the netcore console sample application, and cant seem to find a solution to speed this up.

Then another question, the timeout of the session seems to be 600 seconds by default, do i need to restart the session or is this done by the await method?

Best regards.

AlinMoldovean, 2020-06-03:
Hi @muli1993 ,

You can set a lower value for KeepAliveInterval (default is 5000), but that will cause an overhead due to higher rate of read requests.
```
session.KeepAliveInterval = 1000;
```
_**session.SessionTimeout**_ parameter is the interval used by server to release client sessions without activity

muli1993, 2020-06-05:
Thanks! Sorry for the late answer. 

I have a new problem and I really can't seem to find a solution for it either. How can I find out if the session is still valid ? If i reinit my server, then it will not write values any more since it thinks it is connected. But if I restart the service, which will setup a new connection and monitoring. This will work. I've seen sessionconnected, but it doesnt seem to show anything since the session seems to be there.

Thanks for the answer!

pbakun, 2020-08-07:
Hi,

Check out ConnectServerCtrl in UA Reference Client. You will find keep alive handler which checks if session is still valid:

`private void Session_KeepAlive(Session session, KeepAliveEventArgs e)
        {
            if (this.InvokeRequired)
            {
                this.BeginInvoke(new KeepAliveEventHandler(Session_KeepAlive), session, e);
                return;
            }

            try
            {
                // check for events from discarded sessions.
                if (!Object.ReferenceEquals(session, m_session))
                {
                    return;
                }

                // start reconnect sequence on communication error.
                if (ServiceResult.IsBad(e.Status))
                {
                    if (ReconnectPeriod <= 0)
                    {
                        UpdateStatus(true, e.CurrentTime, "Communication Error ({0})", e.Status);
                        return;
                    }

                    UpdateStatus(true, e.CurrentTime, "Reconnecting in {0}s", ReconnectPeriod);

                    if (m_reconnectHandler == null)
                    {
                        if (m_ReconnectStarting != null)
                        {
                            m_ReconnectStarting(this, e);
                        }

                        m_reconnectHandler = new SessionReconnectHandler();
                        m_reconnectHandler.BeginReconnect(m_session, ReconnectPeriod * 1000, Server_ReconnectComplete);
                    }

                    return;
                }

                // update status.
                UpdateStatus(false, e.CurrentTime, "Connected [{0}]", session.Endpoint.EndpointUrl);

                // raise any additional notifications.
                if (m_KeepAliveComplete != null)
                {
                    m_KeepAliveComplete(this, e);
                }
            }
            catch (Exception exception)
            {
                ClientUtils.HandleException(this.Text, exception);
            }
        }`

Hope it will help!

closed 2020-10-15 by AlinMoldovean