OPCFoundation/UA-.NETStandard #1001: Browse Node in OPC UA server

kamishuki76, 2020-06-01:
Hi. I'm developing an OPC UA simulation server using .NET. I want to create a browse node tab in ServerForm. But I found it in client. Can I use Client Browse Node function to develop a browse node tab in ServerForm?? Please give me some solution. Thank you.

AlinMoldovean, 2020-06-03:
Hi @kamishuki76 ,

Please have a look at #922

kamishuki76, 2020-06-04:
Hi. @AlinMoldovean . Thank you so much. May I ask anthoer question?
Can I Combine the program of boiler Server and reference server? I want to develop a SCADA on this server.

kamishuki76, 2020-06-05:
I have a code to browse node on winform server side like this. 

`public void Initialize(StandardServer server, ApplicationConfiguration configuration)
        {
            m_server = server;
            m_configuration = configuration;
            #region Browse Node
            BrowseDescriptionCollection nodestobrowse = new BrowseDescriptionCollection();
            BrowseDescription nodetobrowse = new BrowseDescription();
           
            nodetobrowse.NodeId = ObjectIds.Server;      
            nodetobrowse.ResultMask = (uint)BrowseResultMask.All;
            nodestobrowse.Add(nodetobrowse);

            OperationContext context = new OperationContext(new RequestHeader(), RequestType.Browse);
            BrowseResultCollection results = null;
            DiagnosticInfoCollection diagnosticInfos = null;

            m_server.CurrentInstance.NodeManager.Browse(
                context,
                null,
                0,
                nodestobrowse,
                out results,
                out diagnosticInfos);
            
            // Display the results
            ReferenceDescriptionCollection nodeReferences = results[0].References;

            ReferenceDescription dt = new ReferenceDescription();
            TreeNode root = new TreeNode();
                             
            foreach (var reference in nodeReferences)
            {
                dt.NodeId = reference.NodeId;
                dt.NodeClass = reference.NodeClass;
                dt.BrowseName = reference.BrowseName;
                dt.DisplayName = reference.DisplayName;
                root.Tag = dt;
                root.Text = reference.NodeId.ToString();
                root.Nodes.Add(new TreeNode());
                BrowseList.Nodes.Add(root);
                root.Expand();
               // BrowseList.SelectedNode = root;

         }                        
            #endregion`
But i when I build and run treeview program. I just display one Node foreach node in serverside 
[image]
So why it is happen? and What should I do??

AlinMoldovean, 2020-06-05:
@kamishuki76 ,

You need to put this part inside _foreach_ loop
```
        dt = new ReferenceDescription();
        root = new TreeNode();
```

kamishuki76, 2020-06-08:
yeah thank. 
`BrowseDescription nodetobrowse1 = new BrowseDescription();
            nodetobrowse1.NodeId = ObjectIds.ObjectsFolder;
            //nodetobrowse1.ReferenceTypeId = 
            nodetobrowse1.BrowseDirection = BrowseDirection.Forward;
            nodetobrowse1.ResultMask = (uint)BrowseResultMask.All;
            nodestobrowse.Add(nodetobrowse1);`
this code just browse all object ID, ex: Server, CTT. Not expand Objects and Variables in 2 Folder. How to add code to expand the Node browsing as Client 
[image]

closed 2020-06-11 by kamishuki76