OPCFoundation/UA-.NETStandard #1004: Creation of subtyped classes (optional Variables)

ThomasRothfuss, 2020-06-05:
I have following Code:

```
    NodeState objectNode = ...

    var alarmNode = new AlarmConditionState(objectNode);
    alarmNode.Create(SystemContext, SystemContext.NodeIdFactory.New(SystemContext, alarmNode), new QualifiedName("Alarm", 1), null, true);
    AddPredefinedNode(SystemContext, alarmNode);
    objectNode.AddChild(alarmNode);

    var warningNode = new AcknowledgeableConditionState(objectNode);
    warningNode.Create(SystemContext, SystemContext.NodeIdFactory.New(SystemContext, warningNode), new QualifiedName("Warning", 1), null, true);
    AddPredefinedNode(SystemContext, warningNode);
    objectNode.AddChild(warningNode);
```

In the tree of the UaExpert there are now two nodes:

```
    |
    |- Alarm
    |   |- AckedState
    |   |   |- Id
    |
    |- Warning
    |   |- AckedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState
    |   |- ConfirmedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState

```
The AlarmConditionState is a subtype of AcknowledgeableConditionState, but its creation leads to a different tree: only the mandatory nodes in the Alarm node exist.

I have extended the code by one line in the class AlarmConditionState (...\Stack\Opc.Ua.Core\Stack\Generated\Opc.Ua.Classes.cs):

```
        #if (!OPCUA_EXCLUDE_InitializationStrings)
        /// <summary>
        /// Initializes the instance.
        /// </summary>
        protected override void Initialize(ISystemContext context)
        {
            base.Initialize(context);       // This is the added line
            Initialize(context, InitializationString);
            InitializeOptionalChildren(context);
        }

```

Then I get following result:

```
    |
    |- Alarm
    |   |- AckedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState
    |   |- ConfirmedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState
    |
    |- Warning
    |   |- AckedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState
    |   |- ConfirmedState
    |   |   |- FalseState
    |   |   |- Id
    |   |   |- TransitionTime
    |   |   |- TrueState

```

The subtrees now look the same.
Without this additional line I have problems creating the optional nodes. This is very cumbersome and error-prone.

Would it be the right approach to add the line?
Then the model compiler should be extended. But I can't assess it.

AlinMoldovean, 2020-11-09:
Waiting for https://github.com/OPCFoundation/UA-ModelCompiler/pull/63 ...

AlinMoldovean, 2021-03-15:
Fix should come in PR #1284

AlinMoldovean, 2021-05-26:
Fixed in #1284

closed 2021-05-26 by AlinMoldovean