OPCFoundation/UA-.NETStandard #1005: How come the BaseObjectState is used to implement Role objects instead of the RoleState?

DmitryFarberov, 2020-06-05:
I would've thought the RoleState class would've been used here, but it doesn't appear to be the case. 

First of all, I would like a conformation that my findings are correct, and they are, I would like to know why it was implemented this way. And if I'm implementing custom Roles, should I use the BaseObjectState or RoleState?

Thank you.

anderZubi, 2021-02-02:
Hello @DmitryFarberov ,
Could you implement custom roles? How did you do it? I'm struggling to figure it out.

Thanks

DmitryFarberov, 2021-02-02:
It's been a while, so let me see if I remember this correctly. It wasn't all that easy or obvious, so here it goes.

First of all, I added a Role Node Manager (all my Node Managers derive from CustomNodeManager2) and provided this override:

        public override void CreateAddressSpace(IDictionary<NodeId, IList<IReference>> externalReferences)
        {
            var accessControl = ConfigurationContext.OpcUaServerConfiguration.AccessControl;
            uint numericNodeId = 100000;

            foreach (var accessControlRole in accessControl.Roles)
            {
                if (accessControlRole.IsWellKnowwn)
                    continue;

                accessControlRole.NodeId = new NodeId(numericNodeId++, NamespaceIndex);

                RoleState role = new RoleState(null)
                                 {
                                     NodeId = accessControlRole.NodeId,
                                     ReferenceTypeId = ReferenceTypeIds.HasComponent,
                                     TypeDefinitionId = ObjectTypeIds.RoleType,
                                     SymbolicName = "ConfiguredRole_" + accessControlRole.Name,
                                     BrowseName = accessControlRole.Name,
                                     DisplayName = accessControlRole.Name,
                                     WriteMask = AttributeWriteMask.None,
                                     UserWriteMask = AttributeWriteMask.None,
                                     EventNotifier = EventNotifiers.None
                                 };

                PropertyState property = role.AddProperty<IdentityMappingRuleTypeCollection>("Identities", DataTypeIds.IdentityMappingRuleType, ValueRanks.OneDimension);
                property.NodeId = new NodeId(numericNodeId++, NamespaceIndex);
                property.ArrayDimensions = new ReadOnlyList<uint>(new List<uint>() {0});
                property.MinimumSamplingInterval = 0;

                property = role.AddProperty<bool>("ApplicationsExclude", DataTypeIds.Boolean, ValueRanks.Scalar);
                property.NodeId = new NodeId(numericNodeId++, NamespaceIndex);
                property.Value = false;
                property.MinimumSamplingInterval = 0;

                property = role.AddProperty<StringCollection>("Applications", DataTypeIds.String, ValueRanks.OneDimension);
                property.NodeId = new NodeId(numericNodeId++, NamespaceIndex);
                property.ArrayDimensions = new ReadOnlyList<uint>(new List<uint>() {0});
                property.MinimumSamplingInterval = 0;

                property = role.AddProperty<bool>("EndpointsExclude", DataTypeIds.Boolean, ValueRanks.Scalar);
                property.NodeId = new NodeId(numericNodeId++, NamespaceIndex);
                property.Value = false;
                property.MinimumSamplingInterval = 0;

                property = role.AddProperty<EndpointTypeCollection>("Endpoints", DataTypeIds.EndpointType, ValueRanks.OneDimension);
                property.NodeId = new NodeId(numericNodeId++, NamespaceIndex);
                property.ArrayDimensions = new ReadOnlyList<uint>(new List<uint>() {0});
                property.MinimumSamplingInterval = 0;

                AddPredefinedNode(SystemContext, role);
            }
        }

Also, in my main Server class derived from StandardServer I provided this override:

        protected override void OnNodeManagerStarted(IServerInternal server)
        {

            // allow base class processing to happen first.
            base.OnNodeManagerStarted(server);

            UapRoleNodeManager roleNodeManager = server.NodeManager?.NodeManagers.FirstOrDefault(m => m is UapRoleNodeManager) as UapRoleNodeManager;
            RoleSetState roleSet = server.ServerObject.ServerCapabilities.RoleSet;

            var accessControl = ConfigurationContext.OpcUaServerConfiguration.AccessControl;
            foreach (var accessControlRole in accessControl.Roles)
            {
                if (accessControlRole.IsWellKnowwn)
                    continue;

                roleSet.AddReference(ReferenceTypeIds.HasComponent, false, accessControlRole.NodeId);
            }

            IList<IReference> references = new List<IReference>();
            roleSet.GetReferences(server.DefaultSystemContext, references);

            foreach (var reference in references)
            {

                NodeId nodeId = new NodeId(reference.TargetId.Identifier, reference.TargetId.NamespaceIndex);

                INodeManager nodeManager;
                NodeHandle handle = server.NodeManager.GetManagerHandle(nodeId, out nodeManager) as NodeHandle;
                string typeName = handle.Node.GetType().Name;
                BaseObjectState roleObject = handle.Node as BaseObjectState;

                IList<IReference> roleReferences = new List<IReference>();
                roleObject.GetReferences(server.DefaultSystemContext, roleReferences);
            }
        }

HTH. Good luck.

EthanChangAED, 2023-07-07:
Converting to discussion.