OPCFoundation/UA-.NETStandard #1007: Implementing OPC UA Server as Windows Background Service

justinWorstCase, 2020-06-08:
I'm implementing an OPC UA server with the .NET standard reference implementation. I'm starting the server (i.e. Server.Run()) from a windows .NET core worker background service (essentially, just a windows service). Everything is running fine except that the client certificate is rejected by the server.

I've manually placed each application instance certificate in the 'trusted' folder once 'rejected'. I've also directly installed them in the CurrentUser\My cert store. 

This certificate is still rejected by the server. Is this a known bug, a mis-configuration, or just not currently possible with the OPC UA stack? Does this mean that, if I want to implement the Opc.Ua.Server stack as a service that I must always set AutoAcceptUntrustedCertificates to true? Any insight you can provide is appreciated.

mregen, 2020-06-08:
Hi @justinWorstCase, could you provide a log file with exception output enabled?

justinWorstCase, 2020-06-08:
So, interestingly enough, no exceptions are logged if I run the server directly as an installed windows service. If I run the worker service from visual studio, the client connects fine and logging does happen as you would except based on the tracing configuration defined in my server.Config.XML.

Here's my current tracing configuration:

```
<TraceConfiguration>    <OutputFilePath>%CommonApplicationData%/SentryOpc/Logs/UAServerLog.txt</OutputFilePath>
    <DeleteOnLoad>true</DeleteOnLoad>
    <!-- Show Only Security, ServiceResultExceptions, Errors and Trace -->
    <TraceMasks>519</TraceMasks>
  </TraceConfiguration>
```

With logging not working in the context of an installed windows service, I tried logging the following details within the event handler: **CertificateValidationEventHandler()**

```
e.Error.StatusCode.ToString();
e.Error.AdditionalInfo.ToString();
e.Error.InnerResult.ToString();
```

Which returned the following output:

**Status code:** BadCertificateUntrusted
**Additional Info:** >>> Certificate is not trusted.
SubjectName: CN=OPC UA Client, C=US, S=Pennsylvania, O=Sentinel Software, DC=localhost
IssuerName: CN=OPC UA Client, C=US, S=Pennsylvania, O=Sentinel Software, DC=localhost

--- at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates) in ...\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 846

--- at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain) in    ...\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 247
 

**Which points to this exception on line 846 of CertificateValidator.cs:**
```
if ((certificateKeyUsage & X509KeyUsageFlags.DataEncipherment) == 0)
{
     throw new ServiceResultException(StatusCodes.BadCertificateUseNotAllowed, "Usage of certificate is not allowed.");
}
```

**And the other points to this statement on line 247 of CertificateValidator.cs:**
`m_validatedCertificates[certificate.Thumbprint]` = new X509Certificate2(certificate.RawData);

mregen, 2020-06-09:
The client certificate not trusted points to a cert issue, the path may only be hit if the cert cannot be found in the trusted store. Are you sure the service can access the cert stores? I'd guess the ACL for the service prevents it from reading/writing config/logs/certs etc.

justinWorstCase, 2020-06-09:
Ok, I added the proper access to the service to rule that out, however this did not fix the main issue. I thought at first that this was the reason that logging was working now, but after reinstalling the service, logging still worked without the command below. I've added a server trace below for your inspection.

For reference, here's the command I used based on:
**(ACE strings and SDDL)**
https://docs.microsoft.com/en-us/windows/win32/secauthz/ace-strings?redirectedfrom=MSDN

`
sc.exe sdset MyServiceName D:(A;;CCLCSWRPWPDTLOCRRC;;;SY)(A;;CCDCLCSWRPWPDTLOCRSDRCWDWO;;;BA)(A;;CCLCSWLOCRRC;;;AU)(A;;CCLCSWRPWPDTLOCRRC;;;PU)(A;;RP;;;IU)
`

Here's the log, the client cert is still being rejected, even though it's in the trusted folder at **[Directory]%LocalApplicationData%/OPC Foundation/pki/rejected**

You'll notice in the log it says that it writes the rejected certificate to the 'rejected' folder, but nothing is actually written here after I've already trusted it. If I delete the cert from the 'trusted' folder it will write it to the 'rejected' folder, but when I move it into the 'trusted' folder, it just repeats the same scenario where the server says the client cert is untrusted yet does not write it to the 'rejected' folder. Not sure how to fix this issue, any advice you can provide is very much appreciated. Thanks for your time!

```
************************* Logging started at 06/09/2020 09:56:15
6/9/2020 09:56:15.567 Checking application instance certificate.
6/9/2020 09:56:15.569 Checking application instance certificate. CN=OPC UA Server, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost.americas.abb.com
6/9/2020 09:56:15.671 Checking domains in certificate. CN=Sentry OPC UA Server, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
6/9/2020 09:56:15.779 The server is starting.
6/9/2020 09:56:16.292 Server: Session Monitor Thread Started.
6/9/2020 09:56:16.295 Server: Publish Subscriptions Thread Started.
6/9/2020 09:56:31.229 Channel 1 in Connecting state.
6/9/2020 09:56:31.230 TCPSERVERCHANNEL SOCKET ATTACHED: 003D893C, ChannelId=1
6/9/2020 09:56:31.267 Channel 2 in Connecting state.
6/9/2020 09:56:31.267 TCPSERVERCHANNEL SOCKET ATTACHED: 0222B57D, ChannelId=2
6/9/2020 09:56:31.272 TCPSERVERCHANNEL ForceChannelFault Socket=0222B57D, ChannelId=0, TokenId=0, Reason=Id: BadConnectionClosed
Description: Remote side closed connection
6/9/2020 09:56:31.273 ***EXCEPTION*** BadConnectionClosed 'Remote side closed connection'
6/9/2020 09:56:31.274 Channel 2 in Faulted state.
6/9/2020 09:56:31.287 Channel 1 in Opening state.
6/9/2020 09:56:31.310 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
6/9/2020 09:56:31.310 Sender Certificate: (none)
6/9/2020 09:56:31.311 Token #0 created. CreatedAt = 13:56:31.311 . Lifetime = 3600000
6/9/2020 09:56:31.316 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.0.0.0] [ID=1] Connected To: opc.tcp://localhost:51210/UA/SentryOpcUa
6/9/2020 09:56:31.317 Token #1 activated. CreatedAt = 13:56:31.311 . Lifetime = 3600000
6/9/2020 09:56:31.317 Channel 1 in Open state.
6/9/2020 09:56:31.318 Channel 1: SendOpenSecureChannelResponse()
6/9/2020 09:56:31.348 Channel 1: ProcessRequestMessage 1
6/9/2020 09:56:31.361 Channel 1: SendResponse 1
6/9/2020 09:56:31.463 TCPSERVERCHANNEL ProcessCloseSecureChannelRequest Socket=TCPSERVERCHANNEL, ChannelId=4032828, TokenId=1
6/9/2020 09:56:31.464 Channel 1 in Closed state.
6/9/2020 09:56:31.465 Channel 1 closed
6/9/2020 09:56:31.497 Channel 3 in Connecting state.
6/9/2020 09:56:31.497 TCPSERVERCHANNEL SOCKET ATTACHED: 03DD3515, ChannelId=3
6/9/2020 09:56:31.497 Channel 3 in Opening state.
6/9/2020 09:56:31.497 Channel 4 in Connecting state.
6/9/2020 09:56:31.497 TCPSERVERCHANNEL SOCKET ATTACHED: 00D8E342, ChannelId=4
6/9/2020 09:56:31.498 TCPSERVERCHANNEL ForceChannelFault Socket=00D8E342, ChannelId=0, TokenId=0, Reason=Id: BadConnectionClosed
Description: Remote side closed connection
6/9/2020 09:56:31.498 ***EXCEPTION*** BadConnectionClosed 'Remote side closed connection'
6/9/2020 09:56:31.498 Channel 4 in Faulted state.
6/9/2020 09:56:31.519 ***EXCEPTION*** BadCertificateUntrusted 'Certificate is not trusted.
SubjectName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
IssuerName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost' Certificate is not trusted.
SubjectName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
IssuerName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
6/9/2020 09:56:31.519 Certificate Vaildation failed for 'CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost'. Reason=BadCertificateUntrusted
6/9/2020 09:56:31.534 Certificate 'CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost' rejected. Reason=BadCertificateUntrusted
6/9/2020 09:56:31.534 Writing rejected certificate to directory: [Directory]%LocalApplicationData%/OPC Foundation/pki/rejected
6/9/2020 09:56:31.540 ***EXCEPTION*** BadCertificateUntrusted ServiceResultException Certificate is not trusted.
SubjectName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
IssuerName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
6/9/2020 09:56:31.540 TCPSERVERCHANNEL ForceChannelFault Socket=03DD3515, ChannelId=0, TokenId=0, Reason=Id: BadSecurityChecksFailed
Description: Certificate is not trusted.
[... cut ...]
SubjectName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost
IssuerName: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost'
6/9/2020 09:56:31.540 Channel 3: SendErrorMessage()
6/9/2020 09:56:31.541 Channel 3 in Faulted state.
6/9/2020 09:56:31.603 TCPSERVERCHANNEL ForceChannelFault Socket=03DD3515, ChannelId=0, TokenId=0, Reason=Id: BadConnectionClosed
Description: Remote side closed connection
6/9/2020 09:56:31.603 ***EXCEPTION*** BadConnectionClosed 'Remote side closed connection'
```

AlinMoldovean, 2020-06-09:
Hi @justinWorstCase ,

The problem seems to be that client certificate does not have the correct _key usage_ set in certificate.Extensions as _X509KeyUsageExtension_.
It needs to have the _DataEncipherment_ flag (i.e. The key can be used for data encryption), otherwise it is not considered valid as an OPC UA application instance certificate.

justinWorstCase, 2020-06-09:
I see, how do I proceed? I’m relying on the self generated application instance certificate, to be clear. I’ve made no adjustments to the certs or the source code of the UA core engine components.

justinWorstCase, 2020-06-09:
I did some more testing based on your comment and found that the DataEncipherment flag is included in the flags as shown below from the output.

I've tracked down the problem to the following condition that is raising the exception when the client tries to connect to the server running as a service:
```
// check if certificate is trusted.
            if (trustedCertificate == null && !isIssuerTrusted)
            {
                if (m_applicationCertificate == null || !Utils.IsEqual(m_applicationCertificate.RawData, certificate.RawData))
                {                  
                    throw ServiceResultException.Create(
                        StatusCodes.BadCertificateUntrusted,
                        "Certificate is not trusted.\r\nSubjectName: {0}\r\nIssuerName: {1}",
                        certificate.SubjectName.Name,
                        certificate.IssuerName.Name);
                }                
            }
```

I logged the properties of the certificates, both 'm_applicationCertificate' and 'certificate' when the condition is met which raises the exception, correct me if I'm wrong, but aren't the byte arrays from the client and server certs always going to be different?
(i.e. the expression:
 !Utils.IsEqual(m_applicationCertificate.RawData, certificate.RawData))
Will always be true here, because we are comparing two different certs.

**[m_applicationCertificate]**
Thumbprint: 33701C5A2D7B08B72085646D5853C1FBAD40B6AD 
Subject: CN=OPC UA Server, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost 
HasPrivateKey: True 
PrivateKey: System.Security.Cryptography.RSACng 
Handle: 2189697130496 
FriendlyName: OPC UA Server 
Issuer: CN=OPC UA Server, C=US, S=Pennsylvania, O=Sentinel Software, DC=localhost 
IssuerName: System.Security.Cryptography.X509Certificates.X500DistinguishedName 
SerialNumber: 46B05186A22A7AB3 
PublicKey: System.Security.Cryptography.X509Certificates.PublicKey 
SignatureAlgorithm: System.Security.Cryptography.Oid
KeyUsage: KeyCertSign, DataEncipherment, KeyEncipherment, NonRepudiation, DigitalSignature 
Version: 3 

**[certificate]**
Thumbprint: 565B791CC518F64A06EE3995EFE1008D9070F49D 
Subject: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost 
HasPrivateKey: False 
PrivateKey:  
Handle: 2189698404736 
FriendlyName:  
Issuer: CN=OPC UA Client, C=US, S=Pennsylvania, O=MyCompanyName, DC=localhost 
IssuerName: System.Security.Cryptography.X509Certificates.X500DistinguishedName 
SerialNumber: 17DA4F9EC06EFFE0 
PublicKey: System.Security.Cryptography.X509Certificates.PublicKey 
SignatureAlgorithm: System.Security.Cryptography.Oid
KeyUsage: KeyCertSign, DataEncipherment, KeyEncipherment, NonRepudiation, DigitalSignature 
Version: 3

justinWorstCase, 2020-06-09:
More findings:

Providing more context:

**When I launch the server from visual studio:**

* Server cert gets place in CurrentUser\My correctly with private key
* Client can connect to it
* Once client connects, client cert is placed in CurrentUser\My correctly with private key

**When I start the server as a windows service:**

* Server cert contains a different thumbprint
* Server cert is automatically placed in rejected folder
* Have to manually install it to place it in the CurrentUser\My store. Once this is done, no private key is available anymore and is lost.
* Client can not connect, client cert is always rejected
* Once server cert is removed from rejected folder, and the service is restarted, it gets placed in the rejected folder again once the service starts, having to be manually installed, which then causes the private key to be lost.

justinWorstCase, 2020-06-11:
I believe I found the root cause of the problem and it comes down to the way %LocalApplicationData% is evaluated at runtime when the server is started as a service. This special folder will evaluate to:

`"C:\Windows\System32\config\systemprofile\AppData\Local"`

Which of course depends on the account used for the service. My requirement is and will be to use the Local System account, so I will need to select a centralized location where the certs are stored that doesn't rely on user accounts.

So, from there I tried to modify the path in the config xml files on client and server side to use a different path from the default for cert store locations, which is:

```
<StorePath>%LocalApplicationData%/OPC Foundation/pki/issuer</StorePath>
<StorePath>%LocalApplicationData%/OPC Foundation/pki/trusted</StorePath>
<StorePath>%LocalApplicationData%/OPC Foundation/pki/rejected</StorePath>
```

I changed to the following:

```
<StorePath>%CommonApplicationData%/SentryOpc/pki/issuer</StorePath>
<StorePath>%CommonApplicationData%/SentryOpc/pki/trusted</StorePath>
<StorePath>%CommonApplicationData%/SentryOpc/pki/rejected</StorePath>
```

Now, regardless of whether the server is started as a service or simply run through as a console application, nothing is ever written to disk and this WARNING shows in the trace log:

`6/10/2020 19:56:25.217 WARNING: Trusted peer store not specified.`

Is it not possible to change the cert location in config file without changing other parts of the code or am I missing something else? Any advice you can provide is much appreciated.

closed 2020-06-11 by justinWorstCase

mregen, 2020-06-12:
Great to hear, thanks for the info. Since the 'legacy' support for running as a service was just removed it would be great if you could share the steps needed. 

Cheers,
Martin

justinConstantTime, 2020-06-22:
@mregen 

There's not much to that part, really. First, to provide context: I am using a .NET core 3.0 worker service that uses **Microsoft.Extensions.Hosting** and **Microsoft.Extensions.Hosting.WindowsServices**.

The worker service model is very simple. Here's the overview of the class:

```
using System.Threading;
using System.Threading.Tasks;
using Microsoft.Extensions.Hosting;

namespace MyServiceNamespace
{
    public class ServiceWorker : BackgroundService
    {
        /// <summary>
        /// The opc server instance
        /// </summary>
        private readonly OpcServer nServer;

        /// <summary>
        /// Default constructor
        /// </summary>
        public ServiceWorker()
        {
            nServer = new OpcServer(autoAccept: false, stopTimeout: 0);
        }

        /// <summary>
        /// The service has started.
        /// </summary>
        /// <param name="cancellationToken">The cancellation token</param>
        /// <returns></returns>
        public override async Task StartAsync(CancellationToken cancellationToken = default)
        {
            //-- start the opc server
            await nServer.RunAsync();
        }

        /// <summary>
        /// The service has stopped.
        /// </summary>
        /// <param name="cancellationToken">The cancellation token</param>
        /// <returns></returns>
        public override async Task StopAsync(CancellationToken cancellationToken = default)
        {
            //-- stop the opc server
            await nServer.StopAsync();
        }

        /// <summary>
        /// The service is executing.
        /// </summary>
        /// <param name="stoppingToken">The cancellation token</param>
        /// <returns></returns>
        protected override async Task ExecuteAsync(CancellationToken stoppingToken)
        {
            //-- while the service is running...
            while (!stoppingToken.IsCancellationRequested)
            {
                await Task.Delay(1000, stoppingToken);
            }
        }
    }
}
```
This service is self-hosted using a console application, so there is no need for any special code to run the OPC UA server beyond what is needed to start any OPC UA server implementation from a windows console application, other than the obvious such as file logging over console writing. I can only speak to what my projected implementation will be, but this service model provides a really nice and clean method to host a windows background service. 

To install the service, (which would appear in the application developer's SETUP.EXE or for testing) you simply publish the application and register the service using the output EXE by means of either a powershell command or sc.exe command. I prefer the latter as it doesn't require any powershell extensions. 

I.E.
```
sc create <ServiceName> displayname=<ServiceDisplayName> binpath=<PathToExe> start="auto"
sc description <ServiceName> <ServiceDescription> 
```

One very useful tool that I've been using when building windows services:
[Windows Process Monitor](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon)

Hope this can be helpful to someone else trying to implement the UA stack using a windows background service!

lanxiaoke, 2024-06-14:
因为证书问题；第三方开的证书是包含完整信息的，自己本地调试做的证书有很多信息不完整；比如加密方式、认证类型等