OPCFoundation/UA-.NETStandard #1008: How to query all nodes from namespace? [question]

STUBA-rupr, 2020-06-08:
Hello,
is there an example of a call a query to return all nodes from specific namespace?
Thank you

PureIso, 2020-06-12:
[Github Repository - ApplicationInstanceManager.cs](https://github.com/PureIso/iso.opc/blob/master/Iso.Opc.ApplicationManager/ApplicationInstanceManager.cs)

You are looking for: GetRootExtendedDataDescriptions()

rudoq32, 2020-06-15:
Hi @PureIso, 
on the end of the day with this :
> Github Repository - ApplicationInstanceManager.cs
> You are looking for: GetRootExtendedDataDescriptions()
I need to call recursively Browse service, right?
[image]

STUBA-rupr, 2020-06-15:
Hi @PureIso, 
on the end of the day with this :
 
> 
> [Github Repository - ApplicationInstanceManager.cs](https://github.com/PureIso/iso.opc/blob/master/Iso.Opc.ApplicationManager/ApplicationInstanceManager.cs)
> 
> You are looking for: GetRootExtendedDataDescriptions()

I need to call recursively Browse service, right?
[image]

PureIso, 2020-06-15:
@STUBA-rupr yes, it has to be recursively called. Unless there is a better way that I am not aware of.

STUBA-rupr, 2020-06-16:
Seems that's it.
FYI, I placed same question in OPC Foundation forum
https://opcfoundation.org/forum/opc-ua-standard/how-to-query-all-nodes-from-namespace-question/

closed 2020-06-16 by STUBA-rupr