OPCFoundation/UA-.NETStandard #1017: browse all node in object Folder in server side

kamishuki76, 2020-06-16:
Hi guy, I have a problem with the server interface. I have a server with the browse nodes in client like this. 
[image]
So i want to all node display in server interface with code
`BrowseDescriptionCollection nodestobrowse = new BrowseDescriptionCollection();

            BrowseDescription nodetobrowse1 = new BrowseDescription();
            nodetobrowse1.NodeId = ObjectIds.ObjectsFolder;
            nodetobrowse1.BrowseDirection = BrowseDirection.Forward;
            nodetobrowse1.IncludeSubtypes = true;
            nodetobrowse1.NodeClassMask = (uint)(NodeClass.Object | NodeClass.Variable);
            nodetobrowse1.ResultMask = (uint)BrowseResultMask.All;
            nodestobrowse.Add(nodetobrowse1);

OperationContext context = new OperationContext(new RequestHeader(), RequestType.Browse);
            BrowseResultCollection results = null;
            DiagnosticInfoCollection diagnosticInfos = null;

            m_server.CurrentInstance.NodeManager.Browse(
                context,
                null,
                0,
                nodestobrowse,
                out results,
                out diagnosticInfos);
            
            // Display the results
            ReferenceDescriptionCollection nodeReferences = results[0].References;

          ReferenceDescription dt = new ReferenceDescription();
            TreeNode root = new TreeNode();

            foreach (var reference in nodeReferences)
           {
               dt = new ReferenceDescription();
                dt = reference;
               root.Nodes.Add(new TreeNode(Utils.Format("{0}", dt)));
                root.Tag = dt;
                root.Text = "Object";                         
            }
            BrowseList.Nodes.Add(root);
            root.Expand();
            BrowseList.SelectedNode = root;`
But it has just CTT and Server. It doesn't expand. 
[image]
Does it have problem with the browse description option??? And what should I do to expand node??? Thanks.

AlinMoldovean, 2020-06-16:
Hi @kamishuki76 ,

You only get the references of the _Objects_ node with your code. 
You need to further browse the _Server_ and _CTT_ nodes.

kamishuki76, 2020-06-19:
Yeah thanks. I've done it.

closed 2020-06-19 by kamishuki76