OPCFoundation/UA-.NETStandard #1022: Copy deploy OPC UA prerequisites

bitbonk, 2020-06-18:
We would like our .NET Core 3.1 OPC UA client application to be copy deployable (just copy the folder to the target machine and run the app). Is there a way bundle everything that is needed for an OPC UA client application (e.g. certificate generator, discovery service ...) with the app, so it not required to run any installers on the target machine?
If so, how?

mregen, 2020-06-18:
`dotnet publish` should do the job to create a folder with all content required to run your app.

bitbonk, 2020-06-18:
@mregen  But don't I need Opc.Ua.DiscoveryServer.exe and Opc.Ua.CerticateGenerator.exe to be present on the target machine? Or isn't that required anymore?

AlinMoldovean, 2020-06-19:
@bitbonk,

`Opc.Ua.CerticateGenerator.exe` is not needed anymore. It was required for [UA-.NET-Legacy](https://github.com/OPCFoundation/UA-.NET-Legacy).
`Opc.Ua.DiscoveryServer.exe` is an optional component.

closed 2020-06-22 by bitbonk