OPCFoundation/UA-.NETStandard #1023: Connect issue from S7 1500 to Reference Server

kamishuki76, 2020-06-19:
Hi guy. Is anyone researching OPC UA Client in Tia Portal???
I'm testing the connection of S7-1500 OPC UA Client (using TIA Portal) and PLCSIM Advanced to Reference Server. 
It's not has certificate when using Simulation PLC. So I choose "No security".  But when I Use "Connect DB" to connect Server, Nothing happens. (ConnectionHdl = 16#0).
[image]
[image]

[image]

Maybe there are 2 situations for this. 
Case 1: Server is busy. Status code is 16#0700_0200.
Case 2: Server is not receive the session from TIA Portal. 
Can you help me Solve this problem??? And how can I make this server is not busy??
Thanks a lot.

FlyinDutchMen, 2020-07-13:
I am using a S7-1500 as well. The PLC is the server and I use C# to create a client. I can connect and all. As soon As I'm messing around with the trusted clients certificate thingy I'm in trouble. SO I think I'm further on then you. If needed you can contact me :)  Please let me know if you still have issues

kamishuki76, 2020-07-16:
Hi. Thank you for contacting me. I have a problem about configuration.
Do you try to change OPC UA Server when it is running?
I've tried to change the config and save this to an config.xml file. But
when I reload the config.xml file, it has error could not load the
configuration file. you can see the issue #1054 to see my code.
I see the error but i can't repair it. Hope you help me to solve this
problem. Thank you

closed 2020-07-26 by kamishuki76

TNAKiet, 2021-05-06:
Hi @kamishuki76 , I meet same problem with you? Did you find the way to fix it? Can you help me?