OPCFoundation/UA-.NETStandard #1025: BadCertificateChainIncomplete

jrboom, 2020-06-19:
I used the Opc.Ua.CertificateGenerator.exe to create a root certificate (CA) and I issued two application certificates with that CA. One for our client application and one for our server. I put the root certificate in the "pki/issuer" folder that our client application references. But when I connect to the server I get an error "BadCertificateChainIncomplete".

Have I configured something wrong?

Thanks,
Joost

mregen, 2020-06-19:
@jrboom to trust the root certificate it must be in the trusted or issuer folder of both client and server, but it sounds as if the server couldn't even find the root cert.

jrboom, 2020-06-22:
The server is a PLC and I added both the application and root certificate. When I debug with the UA Sample Client, I see that the received application certificate contains a reference to the root certificate. The code also detects that the root certificate is a "trusted issuer".

The class `CertificateValidator` creates an `X509Chain` with both the certificates, but the chain doesn't contain any `X509ChainStatus` objects, so the boolean `chainStatusChecked` is not set to `true`.  Therefore it goes into this code:

            if (issuedByCA && (!chainStatusChecked || chainIncomplete))
            {
                throw ServiceResultException.Create(
                    StatusCodes.BadCertificateChainIncomplete,
                    "Certificate chain validation incomplete.\r\nSubjectName: {0}\r\nIssuerName: {1}",
                    certificate.SubjectName.Name,
                    certificate.IssuerName.Name);
            }

AlinMoldovean, 2020-06-22:
Hi @jrboom,

Are you using X509Store type for you certificate stores?

jrboom, 2020-06-22:
When testing with the UA Sample Client, I use the default settings from the repository. There the  application certificate for the client is in the X509Store:
```
    <ApplicationCertificate>
      <StoreType>X509Store</StoreType>
      <StorePath>CurrentUser\UA_MachineDefault</StorePath>
      <SubjectName>CN=UA Sample Client, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
    </ApplicationCertificate>
```
But the trusted issuer certificates are not:
```
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%CommonApplicationData%\OPC Foundation\pki\issuer</StorePath>
    </TrustedIssuerCertificates>
```

AlinMoldovean, 2020-06-23:
@jrboom ,

Can you check if this [commit ](https://github.com/AlinMoldovean/UA-.NETStandard/commit/5d57ae58f26b6f6431f1200bf1602674e4041fca) fixes you issue?

jrboom, 2020-06-24:
@AlinMoldovean Yes, this commit fixes my issue.

commit 5d57ae58 (2020-06-24): Fix for chain status check. #948 Fixes #960  Fixes #996 Fixes #1025
closed 2020-06-24 by AlinMoldovean