OPCFoundation/UA-.NETStandard #1043: StatusValueTimestamp versus Reporting/Sampling behaviour

michiatgithub, 2020-07-03:
Hi dev team
We started a sample application with your .Net Standard Opc Ua library.
It is more or less clear to us how to use it. We just struggle with one issue. We as a OPC UA client want to get notified when just a value or a timestamp in the server as changed (it migth be that we write several times a 5 to a Int16 node). As a customer we want to know when the last write of the 5 was written in the server. We use the sampling MonitoringMode to get only changes of the value, status or timestamp, not periodically. We do not want to get updates periodically, when no value or timestamp has changed.

Steps to reproduce:
1) Our OPC UA Server with a Int16 node (with your .Net library)
2) Our OPC Ua Client with a monitored item (sampling interval = 1sec, queuesize = 1, Filter = StatusValueTimestamp, MonitoringMode = Sampling or Reporting)
3) The sample server application starts writing values to a node 0, 0, 0, 0, 1, 1, 1, 1, 2, 2, 2, 2, and so on and on

Actual Results:
When we use the MonitoringMode.Reporting in the client, we get periodically updates, even if the value, timestamp and status have not changed.
When we use the MonitoringMode.Sampling in the client, we do not get updates, even if the value or the timestamp has changed.

Expected Results:
When we use the MonitoringMode.Sampling, we expect that the client get updates when just the status, timestamp or the value has changed.

Aditional Information:
We write the value as follows. maybe we do that in a wrong way.
<
var variableNode = (VariableNode)_serverImplementation.CurrentInstance.CoreNodeManager.GetLocalNode(new NodeId("TEST_300-WDG-EFB", _namespaceIndex));
variableNode.Value = Convert.ToInt16(value);
>

[OpcUa.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/4869864/OpcUa.zip)

michiatgithub, 2020-07-07:
See also the reference here: https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/

michiatgithub, 2020-07-23:
It seems that the currrent version does not fulfil the behaviour specified. can somebody check and maybe fix that?

AlinMoldovean, 2020-07-27:
@michiatgithub ,

The correct way to implement a server node manager is by extending `CustomNodeManager2`
An example can be found in _Reference Server_ project.

michiatgithub, 2020-08-02:
Does this solve the problem with the periodic node updates when the client subscribes on value, status and timestamp changed?

closed 2020-11-09 by AlinMoldovean