OPCFoundation/UA-.NETStandard #1045: DateTime value

dalekseev-monokot, 2020-07-04:
Hello!

When upgrading the OPC Foundation package version to 1.4.362.42 in our product, I found strange behavior. When I assign an instance of the BaseDataVariableState class a **DateTime** value **(Kind = UTC)** and read it using OPC UA client, I get the value converted to UTC again. 

I checked various nuget packages and found that this occurs starting from version 1.4.360.33. Is this expected behavior or am I missing something?

Thanks.

mregen, 2020-07-06:
Hi @dalekseev-monokot, I guess during this timeframe encoders started to apply the Utils.ToOpcUaUniversalTime() helper for conversion to Utc, as I discovered some inconsistencies how Kind Unspecified is handled in different .Net flavors., 
i) could you double check its actually sending the correct UTC over the wire? 
ii) Could you double check your code if its really handing the DateTime value with kind Utc?
Which .Net or .NET Standard version are you using? Anyway, if there is a bug, I need a repro to add it to the test cases.

Thanks,
Martin

dalekseev-monokot, 2020-07-06:
Hi,

We always use net core 3.1. But after upgrading the OPC UA package version from 1.3.352.12 to 1.4.362.42, our unit tests did not pass. If we install the package version 1.4.359.31, the tests are successful. I can prepare a test project and double-check this behavior, but it will take some time.

--
Best regards,
Dmitriy

mregen, 2020-12-13:
assume the issue is solved, if not please reopen

closed 2020-12-13 by mregen