OPCFoundation/UA-.NETStandard #1046: UaChannelBase.CreateUaBinaryChannel() throws NullRefException if configuration doesn't have a certificate validator

anttu2, 2020-07-06:
While trying to setup some client code to connect to a OPC server using this library, I surprisingly got a `NullReferenceException` thrown when calling `Session.Connect()`.

Found out that the reason was that I hadn't assigned a value to `ApplicationConfiguration.CertificateValidator`, causing the code accessing it in  [UaChannelBase:724](https://github.com/OPCFoundation/UA-.NETStandard/blob/2e47eaa44ff02e2155c42325759a39aff0ec187b/Stack/Opc.Ua.Core/Stack/Client/UaChannelBase.cs#L724) to throw as it assumes that the property is set.

That property should be checked before called but could perhaps also be set to default value in `ApplicationConfiguration`'s constructor (if that makes sense)?

mregen, 2020-07-06:
Hi @anttu2, thanks for the feedback, is this a regression? This code was changed recently.
Anyway, you always need a CertificateValidator of some sort, and it must be configured, so the right fix might be to just throw an ArgumentNullException in Session.Connect.

anttu2, 2020-07-06:
Sorry, forgot the version I'm using, which is the latest stable at the time or writing; v1.4.362.42

stgotte, 2020-07-09:
I would like to connect to a UA Server that doesn't use certificates. Is there a switch to connect without certificates?
The AutoAcceptUntrustedCertificates setting seems to have no effect
[image]

anttu2, 2020-08-28:
@stgotte Maybe you could start a separate ticket for this, as it's not really related to the Null Ref I reported about.
Anyhow, from what I can see in the code that flag only has affect when creating a server, please correct me if I'm wrong @mregen.
You could try fiddle with the `SecurityMode` and `SecurityPolicyUri` settings under `ConfiguredEndpoint.Description`. Don't know if there's any xml config options for those, but at least that seems to make it work for me.

AlinMoldovean, 2020-11-09:
Hi @anttu2 ,

The client can connect to servers that do not expose certificates, but the client application requires a valid application certificate in SecurityConfiguration.

closed 2021-05-26 by AlinMoldovean