OPCFoundation/UA-.NETStandard #1047: How to create an event monitored items ?

Bracewind, 2020-07-06:
Hello everyone !

I am currently using the library to write my own opc ua client/server
I wanted to create a monitored item that looks for events generated by an object. I used for that the CreateMonitoredItem method of the client session. After testing the code, I found out the monitoredItem created was considered as a data change monitored item, and not an event one, meaning that the OnReportEvent method is not correctly set.

As a comment of the CreateMonitoredItems method of the CustomNodeManager2 class, the following remark is written

> ///<remarks>
        /// This method only handles data change subscriptions. Event subscriptions are created by the SDK.
        /// </remarks>

What does it mean ? And how can I specify to the server that my monitoredEvent is looking for events ?

Here is the code of the client : 

            var writeRequestHeader = new RequestHeader();

            //ReadValueId defined in OPC UA spec part 4 7.24, see readOPCUA function in this code for more details
            var fstVariableToRead = new ReadValueId
            {
                NodeId = ExpandedNodeId.ToNodeId(Opc.Ua.Gem.ObjectIds.EquipmentOffline, _session.NamespaceUris),
                AttributeId = Attributes.EventNotifier,
                IndexRange = null,
                DataEncoding = null
            };

            var monitoringConfiguration = new MonitoringParameters()
            {
                ClientHandle = m_maxCurrentHandle,       //id of the monitoring item created
                SamplingInterval = -1,  //-1 means this value must be the same than the one of the subscription
                Filter = null,          //filter specifying in which condition a notification should be sent
                QueueSize = 10,         //length of the queue of notifications
                DiscardOldest = true    //what to do when queue full : replace the oldest value with new value
            };

            var itemsToMonitor = new MonitoredItemCreateRequestCollection();
            var itemToMonitor = new MonitoredItemCreateRequest()
            {
                ItemToMonitor = fstVariableToRead,
                MonitoringMode = MonitoringMode.Reporting, //sample and send notifications to subscriptions
                RequestedParameters = monitoringConfiguration
            };
            itemsToMonitor.Add(itemToMonitor);

            //out parameters
            var valuesRevisedByServer = new MonitoredItemCreateResultCollection();
            var diagnosticErrorCode = new DiagnosticInfoCollection();

            _session.CreateMonitoredItems(writeRequestHeader, subscriptionId,
              TimestampsToReturn.Server,
              itemsToMonitor,
              out valuesRevisedByServer,
              out diagnosticErrorCode
              );
            Console.WriteLine("MonitoringItem created with statusCode : {0}", valuesRevisedByServer[0].StatusCode);

            return valuesRevisedByServer[0].MonitoredItemId;`

AlinMoldovean, 2020-07-06:
@Bracewind,

You need to set a correct filter for the event monitored item and make sure the Object node on the server is configured to report events.
Also, it is not recommended to use "_QueueSize = 10_" for an event monitored item. "QueueSize = 0" is a better option and it means the default queue size used by server.

PureIso, 2020-07-07:
@Bracewind 

[https://github.com/PureIso/iso.opc/blob/master/Examples/Plugins/Iso.Opc.Plugin.InCodeNodeManager/EntryPoint.cs]
(Line 188+ - **Create event state on server side**)

[https://github.com/PureIso/iso.opc/blob/master/Iso.Opc.Core/ApplicationInstanceManager.cs]
(Line 1328+ - **Monitor event and variables**)

Bracewind, 2020-07-07:
@AlinMoldovean @PureIso Thanks a lot for your help, it works perfectly now !!

closed 2020-07-07 by Bracewind