OPCFoundation/UA-.NETStandard #1050: ContentFilter.FilterOperator case insensitive "Like" filtering

11v1, 2020-07-07:
Hi.

 I need to perform event filtering based on event message value. Меssage may be in upper case, in lower case, or it may start from capital. The point is that case should not bother me. I want to receive events depending on the text regadless it's case.
According to "OPC 10000-4 - UA Specification Part 4 - Services 1.04" 7.4.3 FilterOperator Like_6 operator is case sensitive. Is there any way to configure filtering case ignoring (some logical trick)?

AlinMoldovean, 2020-07-08:
Hi @11v1,

In case the number of events generated by server is not too high, one option is to do the filtering on client side with a custom logic at application level.

11v1, 2020-07-08:
Hi @AlinMoldovean.

I am making history event client and server. In my case the number is very high (100000+), so the filtering must be and should be done on server side. 
I have one solution in mind: create custom event type, add property MessageUpper, which will hold the same message but in upper case and in client create requests with where clause to filter by MessageUpper instead of standart Message. But here is one problem: in cases when my history client will be used with 3rd party history server, there will be no my custom property. 
Lake of some "LikeIgnoreCase" seems like omission in UA specification to me , but this is all we have. I am definite that I am not the first who ran in this problem, maybe there is some solution?

mregen, 2021-09-09:
Will need to check spec and may require to open Mantis..

EthanChangAED, 2023-07-25:
Move to discussion.