OPCFoundation/UA-.NETStandard #1051: Publish separate NuGet packages for server and client to slim the dependency tree

anttu2, 2020-07-08:
I was a bit surprised about what dependencies I have to take on using the current NuGet package building a simple OPC UA client.
I'd like to avoid having dependencies on Kestrel etc when working in a purely client based application, assuming that should be possible.

Would it be possible to publish separate packages for Server and Client and then a third meta-package which basically would be like the current one that would have dependencies on both Server and Client NuGets.

This would enable the library users to select packages based on their needs and avoid having transient dependencies on unnecessary packages.

mregen, 2020-07-11:
In full agreement, but we not only need to fix the build process but also the signing process and need to get the approval for it by the owner (OPC Foundation) to release more nuget packages. Please stay tuned, we are working on it, mor thumbs up may help..

mpostol, 2020-07-11:
A similar proposal is in #981. I have pointed out reusability as the main reason to spread functionality to separate NuGet packages.

mregen, 2020-10-11:
Please try this release with seperated nuget packages: https://github.com/OPCFoundation/UA-.NETStandard/releases/tag/1.4.363.107

anttu2, 2020-10-13:
Great stuff! Many thanks!
Just wonder if there's still a requirement to have a dependency on kestrel in the Core library 🤔

MD-V, 2020-10-13:
@anttu2 This dependency exists to support HTTPS OPC UA profiles.

mregen, 2020-10-14:
The next work item is to seperate the Https transport into an optional Nuget package... stay tuned.

mregen, 2020-10-14:
Tracking separation of https in #600...

closed 2020-10-14 by mregen