OPCFoundation/UA-.NETStandard #1054: apply configuration doesn change at runtime

kamishuki76, 2020-07-09:
Hi. I develop a Server Form that can be change the configuration in Server at runtime. I use that code.
m_configuration = configuration;
when I want to change the configuration I use this.
m_configuration.SaveToFile(m_application.ConfigSectionName + "Config.xml"); 
But It's not overwritten the default file Config.xml
So when I restart the server, the server config doesn't change. 
Do I miss something?? If I can't change config.xml file, can I change the configuration.cs of the project?
Please help me solve this. Thank you.

kamishuki76, 2020-07-10:
Hi. I can save this. But another problem is appear. When i use command. 
`#region Private Field
        private StandardServer m_server;
        private ApplicationConfiguration m_configuration;
        private ApplicationInstance m_application;

        public ServerSecurityPolicy Policy { get; private set; }
        #endregion
        #region Constructor
        public EndpointCtrl()
        {
            InitializeComponent();
        }
        #endregion

        #region Public Interface       
        public void Initialize(StandardServer server, ApplicationConfiguration configuration, ApplicationInstance application)
        {
            m_server = server;
            m_configuration = configuration;
            m_application = application;
            ComboBox adlist = new ComboBox();
            foreach (EndpointDescription endpoint in m_server.GetEndpoints())
            {
                if (adlist.FindStringExact(endpoint.EndpointUrl) == -1)
                {
                    adlist.Items.Add(endpoint.EndpointUrl);
                }
            }
            TCP_Adresss_Label.Text = "(" + adlist.Items[0].ToString() + ")";
            HTTPS_Address_Label.Text = "(" + adlist.Items[1].ToString() + ")";
            UA_AddressTB.Text = adlist.Items[0].ToString();
            HTTPS_AddressTB.Text = adlist.Items[1].ToString();
            CheckPolicies(m_configuration);
//           m_configuration.ServerConfiguration.BaseAddresses.
        }
        #endregion

        #region Methods
        private void CheckPolicies(ApplicationConfiguration configuration)
        {
            try
            {
                foreach (ServerSecurityPolicy policy in configuration.ServerConfiguration.SecurityPolicies)
                {
                    if (policy.SecurityMode == MessageSecurityMode.None) UA_SM_NoneCB.Checked = true;
                    if (policy.SecurityMode == MessageSecurityMode.Sign) UA_SM_SignCB.Checked = true;
                    if (policy.SecurityMode == MessageSecurityMode.SignAndEncrypt) UA_SM_Sign_EnCB.Checked = true;
                    if (policy.SecurityPolicyUri == SecurityPolicies.Basic128Rsa15) Basic128Rsa15_CB.Checked = true;
                    if (policy.SecurityPolicyUri == SecurityPolicies.Basic256) Basic256_CB.Checked = true;
                    if (policy.SecurityPolicyUri == SecurityPolicies.Basic256Sha256) Basic256Sha256_CB.Checked = true;
                }
            }
            catch (Exception exception)
            {
                ServerUtils.HandleException(this.Text, exception);
            }
        }
        private void CheckSecurity()
        {
            try
            {

                ServerSecurityPolicy policy = new ServerSecurityPolicy();
                if (UA_SM_NoneCB.Checked)
                {
                    policy = CreateSecurityPolicy(0, 0);
                    Addpolicies(m_configuration, policy);
                }
                else
[... 106 lines cut ...]
        #endregion

        #region Event Hanlder

        #endregion

        private void ApplyBtn_Click(object sender, EventArgs e)
        {
            try
            {
                CheckSecurity(); 
                m_configuration.SaveToFile(m_application.ConfigSectionName + ".Config.xml");
                m_application.LoadApplicationConfiguration(false);
            }
            catch (Exception exception)
            {
                ServerUtils.HandleException(this.Text, exception);
            }
        }
    }
}`
I save this but when I Click apply setting to change the config and reload the server configuration. It has error 
[image]
[image]
Why is this happen??? Is any solution to solve this. Thank you.

mregen, 2022-01-21:
i presume this issue is outdated, please reopen if needed.

closed 2022-01-21 by mregen