OPCFoundation/UA-.NETStandard #1056: InputArguments not connected to MethodState after NodeSet2 model import

geomai, 2020-07-10:
I am trying to import a model from a NodeSet2 xml that was created using a modeling tool. My import function looks the same as the ImportXml function described in #546:

```
private void ImportXml (IDictionary<NodeId, IList<IReference>> externalReferences, string resourcepath)
{
  predefinedNodes = new NodeStateCollection ();

  Stream stream = new FileStream (resourcepath, FileMode.Open);
  Opc.Ua.Export.UANodeSet nodeSet = Opc.Ua.Export.UANodeSet.Read (stream);

  foreach (string namespaceUri in nodeSet.NamespaceUris)
  {
	SystemContext.NamespaceUris.GetIndexOrAppend (namespaceUri);
  }

  nodeSet.Import (SystemContext, predefinedNodes);

  for (int ii = 0; ii < predefinedNodes.Count; ii++)
  {
	try
	{
	  AddPredefinedNode (SystemContext, predefinedNodes[ii]);
	}
	catch (Exception)
	{
	  // ...
	}
  }
  
  AddReverseReferences (externalReferences);
} 
```

While this seems to work in general, I get a BadArgumentsMissing Response from the server whenever I'm trying to call a MethodState with InputArguments. Debugging revealed that the InputArguments on the MethodState always remain null, although they are defined in the NodeSet2 as follows:

```
<UAMethod NodeId="ns=1;s=SomeMethod" BrowseName="1:Some Method">
        <DisplayName>Some Method</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;s=MyInput</Reference>
            <Reference ReferenceType="HasProperty">ns=1;s=MyOutput</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;s=Submodel.A</Reference>
        </References>
    </UAMethod>
    <UAVariable DataType="Argument" ValueRank="1" NodeId="ns=1;s=MyInput" ArrayDimensions="1" BrowseName="InputArguments" ParentNodeId="ns=1;s=SomeMethod" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>InputArguments</DisplayName>
        <Description>InputArguments for method Some Method</Description>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;s=SomeMethod</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ListOfExtensionObject>
                <uax:ExtensionObject>
                    <uax:TypeId>
                        <uax:Identifier>i=297</uax:Identifier>
                    </uax:TypeId>
                    <uax:Body>
                        <uax:Argument>
                            <uax:Name>MyInput</uax:Name>
                            <uax:DataType>
                                <uax:Identifier>i=6</uax:Identifier>
                            </uax:DataType>
                            <uax:ValueRank>-1</uax:ValueRank>
                            <uax:ArrayDimensions/>
                            <uax:Description>
                                <uax:Text>My Input</uax:Text>
                            </uax:Description>
                        </uax:Argument>
                    </uax:Body>
                </uax:ExtensionObject>
            </uax:ListOfExtensionObject>
        </Value>
    </UAVariable>
    <UAVariable DataType="Argument" ValueRank="1" NodeId="ns=1;s=MyOutput" ArrayDimensions="1" BrowseName="OutputArguments" ParentNodeId="ns=1;s=SomeMethod" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>OutputArguments</DisplayName>
        <Description>OutputArguments for method Some Method</Description>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;s=SomeMethod</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ListOfExtensionObject>
                <uax:ExtensionObject>
                    <uax:TypeId>
                        <uax:Identifier>i=297</uax:Identifier>
                    </uax:TypeId>
                    <uax:Body>
                        <uax:Argument>
                            <uax:Name>MyOutput</uax:Name>
                            <uax:DataType>
                                <uax:Identifier>i=6</uax:Identifier>
                            </uax:DataType>
                            <uax:ValueRank>-1</uax:ValueRank>
                            <uax:ArrayDimensions/>
                            <uax:Description>
                                <uax:Text>My Output</uax:Text>
                            </uax:Description>
                        </uax:Argument>
                    </uax:Body>
                </uax:ExtensionObject>
            </uax:ListOfExtensionObject>
        </Value>
    </UAVariable>
```

The error hence is cause by providing one input argument ("MyInput"), however, the server expects none.

Using this NodeSet2 xml in another environment (i. e. another server implementation) did work, though. I also already tried overriding AddBehaviourToPredefinedNode as suggested in #459 - without success. 

I have a feeling my problem relates to the situation described in #546 and #459, yet I was not able to fix my problem on the basis of the provided answers. I would highly appreciate any hints regarding what I may be doing wrong here.

jteaguebec, 2020-10-15:
We're you able to find a solution?

I'm having a similar issue. I am able to route my client method calls by overwritting OnAfterCreate(), however, all the InputArguments that use custom DataTypes always come in as NULL. I am able to successfully write to the OutputArguments which also use custom DataTypes.

mregen, 2022-01-21:
I *think* we need to provide a helper function and samples in the server to do this.

barnstee, 2023-10-04:
> I _think_ we need to provide a helper function and samples in the server to do this.

See here for an example on how to define the input arguments: https://github.com/digitaltwinconsortium/ManufacturingOntologies/blob/2f155d8d2f3875b2267a3c7ed355c8d9bcaca641/Tools/FactorySimulation/Station/StationNodeManager.cs#L129

closed 2025-12-03 by romanett