OPCFoundation/UA-.NETStandard #1058: Question is Server SDK already having implementation for OPC UA History Aggregate functions ?

charlesaugustineabb, 2020-07-13:
Hello,
Question is Regarding OPC UA History Aggregate functions.  OPC Foundation's Dot NET Server SDK having already implementation for History Aggregate functions ,SO that I that i can directly use it in during history server development? or do I  have to implement the History Aggregate functions myself? 
If SDK is having already implementation . Does it implement all 37 HDA aggregates function or how many?

Thanks
charles

AlinMoldovean, 2020-07-13:
Hi @charlesaugustineabb ,

This SDK has implementation examples for the most common aggregate functions:
https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Libraries/Opc.Ua.Server/Aggregates

A Client/Server example project can be found in [HistoricalAccess sample](https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/HistoricalAccess)

charlesaugustineabb, 2020-07-13:
Thank you  AlinMoldovean

closed 2020-10-15 by AlinMoldovean