OPCFoundation/UA-.NETStandard #1062: Self-signed certificates without private key cannot be handled

FlyinDutchMen, 2020-07-14:
When a certificate without a private key is used as the ApplicationCertificate it's not picked up properly. The validation in the method *LoadCertificate* ( class: Session) fails on the find. If the find is set to false the clientCertificate does receive the proper certificate. 

It see this as a bug, because a certificate without a private key is a bit strange but correct. I think there a two ways to tackle this. Either set the Boolean to false or create an additional setting to allow certs without private keys in the ApplicationConfiguration.

    /// <summary>
        /// Load certificate chain for connection.
        /// </summary>
        private static async Task<X509Certificate2> LoadCertificate(ApplicationConfiguration configuration)
        {
            X509Certificate2 clientCertificate;
            if (configuration.SecurityConfiguration.ApplicationCertificate == null)
            {
                throw ServiceResultException.Create(StatusCodes.BadConfigurationError, "ApplicationCertificate must be specified.");
            }

            clientCertificate = await configuration.SecurityConfiguration.ApplicationCertificate.Find(true);

            if (clientCertificate == null)
            {
                throw ServiceResultException.Create(StatusCodes.BadConfigurationError, "ApplicationCertificate cannot be found.");
            }
            return clientCertificate;
        }

mregen, 2020-07-22:
Hi @TimoTielens, could you explain your use case? in general, in OPC UA an app cert without private key doesn't make sense, the private key is needed to validate ownership when negotiating session keys. So without private key, there is no point having an app cert.

gustinmi, 2020-08-07:
For testing purposes please you this unsecure dummy TrustALL approach:

```
// This will catch cert error and trust everything and everybody
config.CertificateValidator.CertificateValidation += new CertificateValidationEventHandler(CertificateValidatorTrustAll);

private void CertificateValidatorTrustAll(CertificateValidator validator, CertificateValidationEventArgs e)
{
    if (e.Error.StatusCode == StatusCodes.BadCertificateUntrusted)
    {
        e.Accept = true; // client_options.AutoAccept;  /// We have here unsecure trust all scenario.
    }
}
```

AlinMoldovean, 2020-08-11:
Private key is mandatory for own ApplicationCertificate .
Certificates without private key can only be used when validating incoming connections.

closed 2020-08-11 by AlinMoldovean