OPCFoundation/UA-.NETStandard #1063: I can't refresh variables

andoniordo, 2020-07-16:
Hello everybody,

I have developed a server that gets the data from a client, let's call it Client1 and I want to read it from a client2. The data transmission is done through an .xml, from which I tell you what variables I want to read.
First, I created the server (node and variable system) and the communication stack for the OPC UA client. Next, I create the socket class (the class that communicates with the machine through proprietary communication, client1) and generates the initial XML that it will send to the machine to know what variables it has to send periodically to the server.
To generate the node and variable system you will use 3 XML documents: Folder_XML, Vars_XML, Method_XML.

The first will be used to generate the file and folder system, giving it the shape we want. The following image will give us the guidelines that will be followed to edit the XML:
[image]

 
The second XML (Vars_XML) indicates the variables that the server will have inside it. The following image will give us the guidelines that will be followed to edit the XML:
 
[image]

At this moment, I manage to read the variables in client2, but for some reason, I don't know that they refresh automatically, although I do see that by manually refreshing the variable it is updated.
As you can see in the image, the value 1 and 2 are different, this is why I have manually updated it.
 

Can someone help me with this problem?
[image]

Thank you very much,
Andoni

Koningsveld, 2020-08-01:
Hi Andoni,

In your server application, where you update the values for your variables.
Do you call the ClearChangeMasks function on the changed variables after updating the values?

andoniordo, 2020-08-04:
Hi!
I suppose where the failure could be is in ReferenceNodeManager and in Reference Server, which is where we read the variables, I have attached a screenshot.

**1.	In ReferenceNodeManager**

private ServiceResult OnWriteValueDiscrete(
            ISystemContext context,
            NodeState node,
            NumericRange indexRange,
            QualifiedName dataEncoding,
            ref object value,
            ref StatusCode statusCode,
            ref DateTime timestamp)
        {
            MultiStateValueDiscreteState variable = node as MultiStateValueDiscreteState;

            TypeInfo typeInfo = TypeInfo.Construct(value);

            if (variable == null ||
                typeInfo == null ||
                typeInfo == Opc.Ua.TypeInfo.Unknown ||
                !TypeInfo.IsNumericType(typeInfo.BuiltInType))
            {
                return StatusCodes.BadTypeMismatch;
            }

            if (indexRange != NumericRange.Empty)
            {
                return StatusCodes.BadIndexRangeInvalid;
            }

            Int32 number = Convert.ToInt32(value);
            if (number >= variable.EnumValues.Value.Length || number < 0)
            {
                return StatusCodes.BadOutOfRange;
            }

            if (!node.SetChildValue(context, BrowseNames.ValueAsText, variable.EnumValues.Value[number].DisplayName, true))
            {
                return StatusCodes.BadOutOfRange;
            }

            node.ClearChangeMasks(context, true);

            return ServiceResult.Good;
        }

**2. In ReferenceNodeManager**

private void DoSimulation(object state)
        {
            try
            {
                lock (Lock)
                {
                    foreach (BaseDataVariableState variable in m_dynamicNodes)
                    {
                        variable.Value = GetNewValue(variable);
                        variable.Timestamp = DateTime.UtcNow;
                        variable.ClearChangeMasks(SystemContext, false);
                    }
                }
            }
            catch (Exception e)
            {
                Utils.Trace(e, "Unexpected error doing simulation.");
            }
        }

**3.	In Reference Server**
protected override void OnServerStopping()
        {
            try
            {
                // check for connected clients
                IList<Session> currentessions = this.ServerInternal.SessionManager.GetSessions();

                if (currentessions.Count > 0)
                {
                    // provide some time for the connected clients to detect the shutdown state.
                    ServerInternal.Status.Value.ShutdownReason = new LocalizedText("en-US", "Application closed.");
                    ServerInternal.Status.Variable.ShutdownReason.Value = new LocalizedText("en-US", "Application closed.");
                    ServerInternal.Status.Value.State = ServerState.Shutdown;
                    ServerInternal.Status.Variable.State.Value = ServerState.Shutdown;
                    ServerInternal.Status.Variable.ClearChangeMasks(ServerInternal.DefaultSystemContext, true);

                    for (uint timeTillShutdown = m_shutdownDelay; timeTillShutdown > 0; timeTillShutdown--)
                    {
                        ServerInternal.Status.Value.SecondsTillShutdown = timeTillShutdown;
                        ServerInternal.Status.Variable.SecondsTillShutdown.Value = timeTillShutdown;
                        ServerInternal.Status.Variable.ClearChangeMasks(ServerInternal.DefaultSystemContext, true);

                        Thread.Sleep(1000);
                    }
                }
            }
            catch
            {
                // ignore error during shutdown procedure.
            }
            base.OnServerStopping();
       }

I will be very grateful if you can help me.

Greetings
Andoni

AlinMoldovean, 2020-11-09:
@andoniordo ,

We could not replicate the issue using the simulation nodes of `ReferenceNodeManager`.
The data change notifications are triggered correctly by Reference Server. e.g. using _Objects\CTT\Scalar\Simulation\Double_ node

closed 2021-03-16 by AlinMoldovean