OPCFoundation/UA-.NETStandard #1066: Read Write Function between PLC S7-1500 OPC UA Client and .Net Server

kamishuki76, 2020-07-23:
Hi. I connect a Real PLC S7 - 1500 as a OPC UA Client to .Net Server Application. 
It's OK when i use Connect FB -> GetNamespaceIndexList FB until the NodeGetHandleList FB -> ReadList or WriteList FB. 
So when I use the readlist and writelist block, It has the error 16#8034_0000 (BadNodeIDUnknown). 
[image]
I tested with a simple SIMATIC OPC UA Server, it can read or write with StatusCode Good. 
Can you investigate the .Net Server sample Application and resolve this?? Thank you
I put link of OPC UA Client FB below:
https://support.industry.siemens.com/cs/document/109762770/s7-user-block-for-the-opc-ua-client-of-a-simatic-s7-1500?dti=0&lc=en-WW.

AlinMoldovean, 2020-07-23:
@kamishuki76 ,

It seems that the OPC UA Client of the PLC is reading some unknown nodes from the server.
That is the only hint we see from the provided error message.

kamishuki76, 2020-07-24:
Yeah. I didn't change the address space and dragged the node from the online server to client many times. But it's not change the statuscode. I check some .Net Server Applications, it is has same error. 
If you have a real PLC, you can test it with them.

AlinMoldovean, 2020-11-09:
This needs to be investigated by the PLC provider.

closed 2021-04-02 by AlinMoldovean

TNAKiet, 2021-05-05:
Hi @AlinMoldovean , I meet same problem with you. Have you find out how to fix it??

TNAKiet, 2021-05-14:
Hi @kamishuki76 , how can you connect PLC to .NET OPC UA Server? Is there anything to pay attention when set up a connection??

dipsiesolutions, 2021-05-22:
@TNAKiet , @kamishuki76 , I got the same issue!! Do you guys have any hint? I can't even establish a connection to the OPCUA .NET Server Application from the PLC S7-1500 OPCUA Client!

kamishuki76, 2021-05-23:
Hi @dipsiesolutions , Do you use real PLC or simulate PLC with PLCSIM Advanced?
- If you connect using your PC as OPCUA Server, you must connect with your PLC with physical LAN network. 
- Pay attention to certificate. If you use simulate PLC, you must connect with No Security.

kamishuki76, 2021-05-23:
Hi @TNAKiet, 
about the connection between PLC and .NET OPC UA Server, please get some notice:
- Your network adapter ( If you use simulate PLC).
- PLC must connect to your LAN.
- Certificate is important. 
- Your server IP, your TIA portal client interface.

thuanvoduy01, 2023-04-28:
Hi @kamishuki76 ,
I'm using PLCSIM Adv and I have connected S7 client to S7 server successfully. Then I replaced S7 server by BoilerServer of OPC Foundation and BatchPlantServer (from [https://www.youtube.com/watch?v=gxA7SDNLHgc&t=1946s](url)). I did try set the `AutoAcceptUntrustedCertificates` in `config.xml` to avoid certificate related problems.
[image]
But still can not established connection to server.
[image]
If you have any idea how to fix this, please let me know. Thanks a lot.

kamishuki76, 2023-05-21:
Hello Mr @thuanvoduy01 . 

Firstly thank you for your interest in this issue.
So the Address Space and Monitor/Simulate tab is the same for the .Net OPC UA Project Reference. So no need to care about it. 

Here are some tips for reconnecting it.
Tips 1: Set an IP network clearly for your server and client. I think it is really needed.
Tips 2: A Client Certificate is needed. TIA Portal has a Certificate management. Cou can research more about it. 
If the "auto accept untrusted certificate" is not working. You can do it manually by finding Client Certificate Location and copying it into the Server Certificate. (and add the Server Certificate to Client configuration on TIA, if you want) Of course, it is temporary for testing. 

Thank you for caring