OPCFoundation/UA-.NETStandard #1072: [Question] Can I use this library to read OPC DA?

captmomo, 2020-07-30:
Hi,

I currently have a working application using this library to read OPC UA tags. Can I use this library for OPC DA tags too?

AlinMoldovean, 2020-07-30:
Hi @captmomo ,

This library can only be used to read over OPC UA protocol.
In order to connect and read from a Classic OPC over OPC DA you need to use a gateway in the middle.
An example of implementation for that can be found in [ComIOP ](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/ComIOP/README.md) sample.
Please note that it is just a sample project that is not fully tested with OPC DA servers from all vendors.

captmomo, 2020-07-31:
thank you! sorry did a search in the wrong project and didn't see it.

closed 2020-07-31 by captmomo