OPCFoundation/UA-.NETStandard #1077: Exception in Serverconfig.xml modification via SecurityConfigurationManager.cs : UpdateDocument : GetObject

GregoireG-C, 2020-08-03:
Hi, i'me new to report Bug,

I have an exception when i try to update a server configuration xml file:
Error at line 1 position 83. Expected element 'SecurityConfiguration' from namespace 'http://opcfoundation.org/UA/SDK/Configuration.xsd' .. Detection of 'Element' with the name 'ApplicationCertificate', namespace 'http://opcfoundation.org/UA/SDK/Configuration.xsd'. "} System.Exception {System.Runtime.Serialization.SerializationException

I manage to solve it changing the GetObject method.

Explanation:
innerXml will send the child node name. 
ReadObject will check whether the name of the object corresponds to the root name value supplied in the constructor.

In the case we use GetObject on a node which is not the root node (using as a relative father node), the name will not match and ReadObject will send an exception.
So I think we should use OuterXml ; using it we will take back the whole branch from the root node to the children nodes.

Hope, my explanations are clear U_U.

I'me not used to the github community colaboration. I have proposed a changement on the master branch on this matter.

Regards

GregoireG-C, 2020-08-03:
[https://github.com/OPCFoundation/UA-.NETStandard/commit/de3b11c1609dddeb5ebfcd83590ca3680dd1ed55](url)
My modification proposal

GregoireG-C, 2020-08-04:
An other issue : There is a access violation on the XML document in WriteConfiguration method.
After loading the xml file, we have to close the fileStream if we want to write on it later.

Regards

mrsuciu, 2021-09-09:
Sorry for the late response but can you initiate a Pull Request with your proposed changes ?

closed 2022-07-07 by GregoireG-C