OPCFoundation/UA-.NETStandard #1078: System.sys still working ? (instead of Kestrel)

GregoireG-C, 2020-08-03:
Hi guys,

I'me currently working on an update of the stack from an old version to the 1.4.362.42.
And I can see that there are a lot of changement on the web server. You made an access to **Kestrel** allowing us to make the stack usable on Linux (I guess) and I was wondering if we could still use **System.sys**. It will be quite hard for me to merge to kestrel for backward compatibility.
**ASP.NET** Core allows the use of **system.sys** but I wonder if the stack has implemented it.
If the stack work with system, can you explain me (or maybe give me an example) on  how I should use the stack.
 
Another thing ; I woud like to avoid using  .NET Standard 2.0. Is it possible to make the stack works just with .Net Framework 4.6.2. ?

I hope, I'me clear on my demand. I'me new on it and it's the first time for me to ask for help. 

Can someone help me with this problem?
Thanks for your time :D

AlinMoldovean, 2020-08-11:
Hi @GregoireG-C ,

This stack does not have support and implementation for _HTTP.sys_ (because it only runs on Windows).
In order to use HTTP.sys you need to implement a custom _ITransportListener_ based on it.

For .NET Standard vs .NET Framework question, the stack projects produce separate dlls in "netstandard2.0" and "net462" output  folders so you can chose to use any of the required libraries.

GregoireG-C, 2020-08-24:
I hva an other question, is it possible to create several endpoints with the same port ?

closed 2020-09-23 by GregoireG-C