OPCFoundation/UA-.NETStandard #1080: difference between RegisterNodeManager and RegisterNamespaceManager of MasterNodeManager class [Question]

STUBA-rupr, 2020-08-04:
I am having an issue with dynamically added namespace.
I can add a namespace by overriding  MasterNodeManager.CreateMasterNodeManager as 
```
// create the custom node managers.
nodeManager = new ServerTypeNodeManager(server, configuration);            
nodeManagers.Add(nodeManager);

 // create master node manager.
masterNodeManager = new MasterNodeManager(server, configuration, null, nodeManagers.ToArray());
```
Now I can add a new node to nodeManager without issue.

But once I add a new namespace by MasterNodeManager.RegisterNamespaceManager:
```
ServerTypeNodeManager nodeManager;
nodeManager = new ServerTypeNodeManager(pserver.CurrentInstance, uri);

// this adds node manager to private MasterNodeManager.m_namespaceManagers
pserver.masterNodeManager.RegisterNamespaceManager(uri, nodeManager);

int index = pserver.CurrentInstance.NamespaceUris.GetIndex(uri);

// add node "managerIdentifier" to be able to retrieve manager handle
NodeId nodeId = new NodeId("ns=" + index + ";s=managerIdentifier");
nodeManager.AddNode(null, "managerIdentifier", "managerIdentifier", null, (ushort)index, NodeClass.Object);
```
I have an issue like this:
[image]

Am I doing something wrong, RegisterNamespaceManager is not for the pusrpouse I am trying to use it?

AlinMoldovean, 2020-08-11:
Hi @STUBA-rupr ,

Please make sure to implement the _GetManagerHandle()_ and _ValidateNode()_ overrides in your _ServerTypeNodeManager_ class

STUBA-rupr, 2020-08-12:
Hello @AlinMoldovean,
the _GetManagerHandle()_ returns the correct manager handle.
(in main)
```
ushort ns1 = aggregatingServer.AddNamespaceUri("ns1:urn:server.application.IOT2040");

            //ns1 = 2;

            // Create server object 
            //Opc.Ua.NodeId.Create(Opc.Ua.ObjectTypes.ServerType, Opc.Ua.Namespaces.OpcUa, namespaceUris)
            NodeId parentNodeId =   ObjectIds.ObjectsFolder;
            NodeId newNodeId = new NodeId("ns=" + ns1 +";s=SERVER:XXXXX");

            BaseInstanceState baseInstanceState = aggregatingServer.AddNode(parentNodeId,
                (string)newNodeId.Identifier,
                (string)newNodeId.Identifier, ObjectIds.ObjectsFolder, ns1, NodeClass.Object);
```

(in class AggregatingServer):
```
public BaseInstanceState AddNode(NodeId parentNodeId, string path, string name, NodeId nodeType, ushort namespaceIndex, NodeClass nodeClass = NodeClass.Unspecified)
        {
            Dictionary<NodeId, IList<IReference>> externalReferences = new Dictionary<NodeId, IList<IReference>>();

            ServerType serverType = (ServerType)server;
            ServerTypeNodeManager serverTypeNodeManager = null;

            // get approprite node manager
            INodeManager nodemanager = null;
            NodeId nodeId = new NodeId("ns=" + namespaceIndex + ";s=managerIdentifier");

            
            serverType.masterNodeManager.GetManagerHandle(nodeId, out nodemanager);
            if (nodemanager.GetType() == typeof(ServerTypeNodeManager))
            {
                serverTypeNodeManager = nodemanager as ServerTypeNodeManager;
            }
            
            // override with serverTypeNodeManager ns=2
            //serverTypeNodeManager = serverType.nodeManager;

            // in which noide manager ? maybe try in all
            NodeState parentNode = serverTypeNodeManager.FindNodeInAddressSpace(parentNodeId);

            // Create Folder for Aggregated servers
            BaseInstanceState root = serverTypeNodeManager.AddNode(parentNode,
                    path, name, nodeType, namespaceIndex, nodeClass);

            // link root to objects folder.
            IList<IReference> references = null;
            if (!externalReferences.TryGetValue(parentNodeId, out references))
            {
                externalReferences[parentNodeId] = references = new List<IReference>();
            }
            references.Add(new NodeStateReference(parentNodeId, false, root.NodeId));

            ((ServerType)server).AddExternalRefferences(externalReferences);

            return root;

        }
```
ServerTypeNodeManager.AddNode does this:
```
public BaseInstanceState AddNode(NodeState parent, string path, string name, NodeId nodeTypeId, ushort namespaceIndex, NodeClass nodeClass = NodeClass.Unspecified)
        {
            BaseInstanceState baseInstanceState;
            switch (nodeClass)
            {
                
                case NodeClass.Variable:
                case NodeClass.VariableType:

                    if((uint)nodeTypeId.Identifier == (uint)VariableTypeIds.PropertyType.Identifier)
                        baseInstanceState = new PropertyState(null);
                    else
                        baseInstanceState = new BaseDataVariableState(null);
                    
                    break;
                case NodeClass.Method:
                    MethodState methodState = new MethodState(null);
                    baseInstanceState = methodState;                                       
                    break;

                case NodeClass.Object:
                case NodeClass.ObjectType:
                default:
                    baseInstanceState = new BaseObjectState(null);
                    break;
            }            

            
            baseInstanceState.Create(SystemContext, new NodeId(path, namespaceIndex), new QualifiedName(path, namespaceIndex), name, true);            

            if (parent != null)
            {
                parent.AddChild(baseInstanceState);
                baseInstanceState.AddReference(ReferenceTypes.Organizes, true, parent.NodeId);
            }

            //baseInstanceState.EventNotifier = EventNotifiers.SubscribeToEvents;
            
            //references.Add(new NodeStateReference(ReferenceTypes.Organizes, false, root.NodeId));

            AddRootNotifier(baseInstanceState);
            AddPredefinedNode(SystemContext, baseInstanceState);

            return baseInstanceState;
        }
```
The ValidateNode is not available for override.

AlinMoldovean, 2021-06-09:
The missing part was adding the NodeManager in m_nodeManagers:

`masterNodeManager.NodeManagers.Add(nodeManager);`

closed 2021-06-09 by AlinMoldovean