OPCFoundation/UA-.NETStandard #1086: Reference server and СTT Testing

inspirit1986, 2020-08-11:
Hi, as I can see https://github.com/OPCFoundation/UA-.NETStandard/releases

1.4.362.42 version are tested with OPC UA Compliance Test Tool 1.04.9.396 so,
Does UAReferenceServer.ctt.xml in Application is suitable for ctt testing with ctt (1.04.9.396) and what profiles should I use for correct ctt testing? Standard 2017 UA Server Profile or Standard UA Server Profile?

Sorry for stupid questions :)

AlinMoldovean, 2020-08-11:
Hi @inspirit1986 ,

The _UAReferenceServer.ctt.xml_ project can be used, but it needs some local changes, e.g. ServerUrl, certificates, NodeIds.
The supported and tested profiles are: 

- Standard 2017 UA Server
- Data Access Server Facet
- Method Server Facet

inspirit1986, 2020-08-14:
@AlinMoldovean Hi, thank you for reply.
At this moment I was built Reference server 1.4.362.42 and test it with ctt(1.04.9.396) with Standard 2017 UA Server,Data Access Server Facet and Method Server Facet.

I used the latest version of the UAReferenceServer.ctt.xml from master and I have a couple of failed(red) test in ctt.
Such as: 

1. Attribute Services/Attribute Write Values/003.js 
[image]
2. Base Information/Base Info Core structure/001.js (A lot of errors in this case..)
[image]
3. Monitored Item Services/Monitor Items Deadband Filter/[14,15,16,18].js
[image]
4. Subscription Services/Subscription Basic/052.js
[image]

Can you explain, what I'm doing wrong?
I think my ctt project settings(such as nodeID's settings) or maybe Reference Server config isn't correct enough.
Thank in advance.

AlinMoldovean, 2020-10-30:
Hi @inspirit1986 ,

I checked this.

Point 1) is caused by a null value written by CTT when trying to simulate a random vale for ByteString node.
This should be fixed on both CTT and Server side

Point 2) is caused by a wrong structure of _ServerRedundancy_ node. The source of the issue is inside the NodeSet imported from _UA-Nodeset_ repository

Point 3) and 4) - the tests fail only sometimes. If the tests are run individually they pass. This is acceptable.

AlinMoldovean, 2020-11-09:
Point 2) should be fixed when a new NodeSet with a fix is published. Issue reported in Mantis 6215.

AlinMoldovean, 2021-03-15:
Waiting for a CTT version containing the fix from Mantis 6341...

AlinMoldovean, 2021-04-02:
Point 2) fixed in #1284.
Only Point 1) remains an open issue.

mregen, 2021-07-15:
@AlinMoldovean is there an update for point 1)?

AlinMoldovean, 2021-07-15:
Hi @mregen ,
 
For point 1) The CTT test was fixed according to Mantis 6341, but an official UA CTT build containing the fix was not yet published.
For a better handling of the _null_ values in `BaseVariableState ` class an API breaking change is required.

AlinMoldovean, 2021-07-16:
@mregen,

A new CTT version 1.04.9.398 containing the fix for point 1) was released.
I run the tests and _"Attribute Write Values\003.js"_ is passing now.
The issue can be closed now.

[image]

closed 2021-07-16 by AlinMoldovean