OPCFoundation/UA-.NETStandard #1088: Need correct steps to setup OPC UA Server and OPC UA Client with Reverse connection.

aravinds4uAll, 2020-08-11:
Hi friends,

Can you please guide me the correct steps/sequence to run an OPC UA Server with reverse connection and connect an OPC UA Client with reverse connection. I badly need it for my work to complete. Please guide me on which project to build and run first followed by the other. Basically the sequence. Also please do guide me on any hacks or extra steps to be taken care for reverse connection to work.

Thanks & Regards
Aravind S.

aravinds4uAll, 2020-08-11:
Please also guide me on what exactly needs to be passed to the reverse connection manager uri. and also the Server uri. I am getting confusion over this in the OPC Client with reverse connection.

mregen, 2020-08-12:
I suggest to wait for #1073 and then use the Reference server as a reverse connect server sample. For the client side we need to look for a good sample code in this repo, as you already noticed the nuget package support is broken due to an exception #1089 .

closed 2020-09-01 by mregen