OPCFoundation/UA-.NETStandard #1089: Getting 'TransportListenerBindings' exception on trying to do OPC Reverse connect from OPC UA Client program.

aravinds4uAll, 2020-08-11:
1. I have made up the OPC UA Server with Reverse Connect. (Matrikon Dispatcher). It is able to do reverse connection with the Matrikon dispatcher client tool.
2.  Tried the same OPC Client scenario with ComplexClient project in the client machine but met with the below error.
3. Please let me know what should be set for the ReverseConnectUri.

            ReverseConnectManager reverseConnectManager = null;
            if (ReverseConnectUri != null)
            {
                // start the reverse connection manager
                reverseConnectManager = new ReverseConnectManager();
                reverseConnectManager.AddEndpoint(**ReverseConnectUri**);
                reverseConnectManager.StartService(config);
            }

Getting the below TransportListenerBindings exception at OPC UA Client with reverse connection.

opc.tcp://10.157.55.140:56000System.AggregateException: One or more errors occurred. (The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception.) ---> System.TypeInitializationException: The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at Opc.Ua.Bindings.TcpListener.Dispose(Boolean disposing)
   at Opc.Ua.Bindings.TransportListenerBindings.IsTransportListenerType(Type systemType)
   at System.Linq.Enumerable.WhereArrayIterator`1.MoveNext()
   at Opc.Ua.Bindings.TransportListenerBindings..cctor()
   --- End of inner exception stack trace ---
   at Opc.Ua.Bindings.TransportListenerBindings.GetTransportListener(String uriScheme)
   at Opc.Ua.ReverseConnectHost.CreateListener(Uri url, EventHandler`1 OnConnectionWaiting, EventHandler`1 OnConnectionStatusChanged)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpointInternal(Uri endpointUrl, Boolean configEntry)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpoint(Uri endpointUrl)
   at NetCoreConsoleClient.MySampleClient.ConsoleSampleClient() in C:\Users\Administrator\source\repos\NetCoreComplexClient\Program.cs:line 279
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at NetCoreConsoleClient.MySampleClient.Run() in C:\Users\Administrator\source\repos\NetCoreComplexClient\Program.cs:line 203
---> (Inner Exception #0) System.TypeInitializationException: The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at Opc.Ua.Bindings.TcpListener.Dispose(Boolean disposing)
   at Opc.Ua.Bindings.TransportListenerBindings.IsTransportListenerType(Type systemType)
   at System.Linq.Enumerable.WhereArrayIterator`1.MoveNext()
   at Opc.Ua.Bindings.TransportListenerBindings..cctor()
   --- End of inner exception stack trace ---
   at Opc.Ua.Bindings.TransportListenerBindings.GetTransportListener(String uriScheme)
   at Opc.Ua.ReverseConnectHost.CreateListener(Uri url, EventHandler`1 OnConnectionWaiting, EventHandler`1 OnConnectionStatusChanged)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpointInternal(Uri endpointUrl, Boolean configEntry)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpoint(Uri endpointUrl)
   at NetCoreConsoleClient.MySampleClient.ConsoleSampleClient() in C:\Users\Administrator\source\repos\NetCoreComplexClient\Program.cs:line 279<---
opc.tcp://10.157.55.140:56000System.AggregateException: One or more errors occurred. (The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception.) ---> System.TypeInitializationException: The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at Opc.Ua.Bindings.TcpListener.Dispose(Boolean disposing)
   at Opc.Ua.Bindings.TransportListenerBindings.IsTransportListenerType(Type systemType)
   at System.Linq.Enumerable.WhereArrayIterator`1.MoveNext()
   at Opc.Ua.Bindings.TransportListenerBindings..cctor()
   --- End of inner exception stack trace ---
   at Opc.Ua.Bindings.TransportListenerBindings.GetTransportListener(String uriScheme)
   at Opc.Ua.ReverseConnectHost.CreateListener(Uri url, EventHandler`1 OnConnectionWaiting, EventHandler`1 OnConnectionStatusChanged)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpointInternal(Uri endpointUrl, Boolean configEntry)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpoint(Uri endpointUrl)
   at NetCoreConsoleClient.MySampleClient.ConsoleSampleClient()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at NetCoreConsoleClient.MySampleClient.Run()
---> (Inner Exception #0) System.TypeInitializationException: The type initializer for 'Opc.Ua.Bindings.TransportListenerBindings' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at Opc.Ua.Bindings.TcpListener.Dispose(Boolean disposing)
   at Opc.Ua.Bindings.TransportListenerBindings.IsTransportListenerType(Type systemType)
   at System.Linq.Enumerable.WhereArrayIterator`1.MoveNext()
   at Opc.Ua.Bindings.TransportListenerBindings..cctor()
   --- End of inner exception stack trace ---
   at Opc.Ua.Bindings.TransportListenerBindings.GetTransportListener(String uriScheme)
   at Opc.Ua.ReverseConnectHost.CreateListener(Uri url, EventHandler`1 OnConnectionWaiting, EventHandler`1 OnConnectionStatusChanged)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpointInternal(Uri endpointUrl, Boolean configEntry)
   at Opc.Ua.Client.ReverseConnectManager.AddEndpoint(Uri endpointUrl)
   at NetCoreConsoleClient.MySampleClient.ConsoleSampleClient()<---
opc.tcp://10.157.55.140:56000

mregen, 2020-08-12:
Thats a known issue and already fixed in the stack, but will need a nuget release for the samples repo.

aravinds4uAll, 2020-09-29:
I am still getting issue with the latest nuget package also. Anybody getting similar issue. Stack code is working fine,

mregen, 2020-10-02:
@aravinds4uAll please let me know if it works for you now: https://github.com/OPCFoundation/UA-.NETStandard-Samples/pull/52

mregen, 2020-10-06:
Please reopen if there is still an issue

closed 2020-10-06 by mregen

aravinds4uAll, 2020-10-06:
@mregen Sorry, I got bit busy with other work. I will revisit it and recheck it once I am done with it. Please excuse me until then.