OPCFoundation/UA-.NETStandard #1090: Is the OPC UA Revese Connect functionality fully operational?

aravinds4uAll, 2020-08-11:
Hi

Please don't mind me asking. I see lot of check-ins going on in the Reverse Connect part. As I am fully relying o your latest code and nuget packages for my development work. Kindly let me know whether the support for reverse connect is fully done or is it still in development.

Please also guide me in 4-5 steps on how to setup the OPC UA Server and OPC UA Client reverse connection. That will be helpful.

Warm Regards
Aarvind

mregen, 2020-08-12:
Hi @aravinds4uAll, the reverse connect functionality is working in the nuget release. But it will be much better in the next nuget release.
The current release works, but has some limitations as for example it allows only a single connection to a server and the client did not properly handle the reconnect case. Also reverse hello messages were not held open, so it always takes a long time to connect on demand. But most improvements are on the client side and are tested with the Aggregation server in the samples repo. When #1073 is fully integrated I'd consider it stable.

aravinds4uAll, 2020-08-12:
> Hi @aravinds4uAll, the reverse connect functionality is working in the nuget release. But it will be much better in the next nuget release.
> The current release works, but has some limitations as for [...]

aravinds4uAll, 2020-08-12:
Sorry, Closed by mistake.

closed 2020-08-13 by mregen