OPCFoundation/UA-.NETStandard #1091: Session.Read doesn't seems to timeout

panManfredini, 2020-08-12:
I'm hitting a problem with the Client Session API. 
When the connection is lost with the server, and one calls Session.Read the method does not time out, it does not follow "OperationTimeout" and ignores RequestHeader::TimeoutHint, it just hangs forever. Is this an expected behavior? And if so is there a way I can work around it?

Thanks a lot! 
best regards,

            Alessandro.

mrsuciu, 2021-09-09:
Sorry for the delayed response. Is this still an issue and if yes can you provide an example ?

mregen, 2022-01-12:
should somehow time out, we need to check this

romanett, 2025-05-03:
I cannot reproduce this with the current master, the Async Operation is started with the Operation Timrout of the channel and is cancelled with BadRequestTimeout after this time elapsed.

https://github.com/OPCFoundation/UA-.NETStandard/blob/d8b2087ff881b40db945ca235a083f591a11b094/Stack/Opc.Ua.Core/Stack/Tcp/ChannelAsyncOperation.cs#L29-L40

closed 2025-05-03 by romanett