OPCFoundation/UA-.NETStandard #1094: [Data Access Server]

jprestubog, 2020-08-17:
In the image below you will noticed that the "DVCL_JPLR.Health" source timestamp does not changed unlike all the other tags being monitored. 

[image]

Every minute, all of the monitored tags are being updated and I am sure that all values are being saved. I verify that this statement is true because it reflected in the database and also dragging again the "DVCL_JPLR.Health" in the Data Access View (another entry) will reflect the latest timestamp that is in the database see image below. It just so happen that previous value is the same with the new value.

[image]

In my observation, it seems like if you updated all the values of the monitored tags but the value is the same as previous value the Source Timestamp does not changed. To test this, I appended the EXPECTED Source Timestamp to force the value of monitored tag to vary every minute (see image beloe)

[image]

Is this how do we expect the behavior of the OPC UA-.Net Framework? if not, how and where to change this?

mrsuciu, 2021-09-09:
Sorry for the late answer. Can you provide some reproduction steps ? What I suppose is that the Server does not update the SourceTime stamps. Could it be so ? If so please contact the server vendor for this issue.

mregen, 2022-01-21:
There are too many reasons to decide whats being wrong, please reopen if necessary.

closed 2022-01-21 by mregen