OPCFoundation/UA-.NETStandard #1102: ApplicationInstance.CheckApplicationInstanceCertificate side effect

jordan-gullen-nextdc, 2020-08-26:
Hi team,

I believe the implementation of ApplicationInstance.CheckApplicationInstanceCertificate should be changed.

The function has a side effect of deleting your configured certificate and creating a new one if the configured certificate was found to be invalid. This was a problem for me as the deletion and replacement consider the check to return true leaving the user completely unaware of this side effect. The other issue is the certificate that it creates mimics the subject name of the original certificate and is self-signed which makes it hard to identify this swap happened and still results in a connection problem as the certificate is not trusted.

Please update this function to only check the certificate with no side effects.

mregen, 2021-01-15:
Hi @jordan-gullen-nextdc, agreed, this is an issue especially when there was a signed certificate.

mregen, 2021-09-09:
Imho we should never update an existing cert which just expired. Also never update a CA issued cert.

mrsuciu, 2021-09-10:
We need to check if the certificate error is suppressible and let it continue if so and skip generating a new one.
It might be that the mechanism which deletes the "invalid" certificate kicks in and triggers this behavior.

PR #1525 (2021-09-28): Application config improvements for cert recreation
- do not silent recreate a certificate if a matching cert subject is available, enforce manual deletion or replacement
- allow the application cert to be used when expired or not yet valid
- warn/assert if an app cert is loaded without loading the private key 
fixes #1162 , fixes #1102
commit e5baad9c (2021-09-28): Application config improvements for cert recreation (#1525)

- do not silent recreate a certificate if a matching cert subject is available, enforce manual deletion or replacement
- allow the application cert to be used when expired or not yet valid
- warn in trace if an app cert is loaded without loading the private key 
fixes #1162 , fixes #1102
closed 2021-09-28 by mregen