OPCFoundation/UA-.NETStandard #1104: Is it possible to get the value change of the subscribed tags through Sunscription FastDataChangeNotificationEventHandler

aravinds4uAll, 2020-08-28:
For my developemnt work, I wanted to subscribe tags in a subscription and get the notification for the value change. 
My initial plan was to add handler to the Monitoreditems list but that creates lot of even handlers waiting for a value change.

Instead of that, Can we achive it using Sunscription FastDataChangeNotificationEventHandler? 

Saw this code in the .NET Standrard. Somebody can share more details on this will be great.

  public void SetSubscription(Subscription subscription)
        {
            if (m_subscription != null)
            {
                m_subscription.PublishStatusChanged -= m_PublishStatusChanged;
                m_subscription.FastDataChangeCallback = null;
                m_subscription = null;
            }

            m_session = null;
            m_subscription = subscription;
            m_subscription.DisableMonitoredItemCache = true;
            m_subscription.PublishStatusChanged += m_PublishStatusChanged;
            **m_subscription.FastDataChangeCallback = OnDataChange;**
            m_dataset.Tables[0].Rows.Clear();

            if (m_subscription != null)
            {
                m_session = subscription.Session;
                m_subscription.Handle = this;
            }
        }

 private void OnDataChange(Subscription subscription, DataChangeNotification notification, IList<string> stringTable)
        {
            if (!Object.ReferenceEquals(subscription, m_subscription))
            {
                return;
            }

            if (this.InvokeRequired)
            {
                this.BeginInvoke(new FastDataChangeNotificationEventHandler(OnDataChange), subscription, notification, stringTable);
                return;
            }

            try
            {
                foreach (MonitoredItemNotification itemNotification in notification.MonitoredItems)
                {
                    MonitoredItem monitoredItem = subscription.FindItemByClientHandle(itemNotification.ClientHandle);

                    if (monitoredItem == null)
                    {
                        continue;
                    }

                    DataRow row = (DataRow)monitoredItem.Handle;

                    if (row.RowState == DataRowState.Detached)
                    {
                        continue;
                    }

                    UpdateRow(row, itemNotification);

                    if (m_EditComplexValueDlg != null && Object.ReferenceEquals(m_EditComplexValueDlg.Tag, monitoredItem))
                    {
                        m_EditComplexValueDlg.UpdateValue(monitoredItem.ResolvedNodeId, monitoredItem.AttributeId, null, itemNotification.Value.Value);
                    }
                }
            }
            catch (Exception exception)
            {
                ClientUtils.HandleException(this.Text, exception);
            }
        }

MD-V, 2020-08-31:
Hello @aravinds4uAll,

you can use the FastDataChangeCallback event handler for your use case. All information that you need is in the DataChangeNotification notification parameter of the event.

aravinds4uAll, 2020-09-29:
Implemented it successfully. Thank you.

closed 2020-09-29 by aravinds4uAll