OPCFoundation/UA-.NETStandard #1112: Unloading broken assembly Opc.Ua.Core.dll in Unity

fejesgergorichard, 2020-09-09:
Hi,
I am currently trying to develop an OPC UA Client for HoloLens 2 in Unity 2019.2.21.  I downloaded the UA-.NETStandard nuget package and dragged the Opc.Ua.Client, Opc.Ua.Configuration and Opc.Ua.Core dll-s with a link.xml into the Assets/Plugins folder.
I am constantly getting this error message in Unity editor: "Unloading broken assembly Assets/Plugins/Opc.Ua.Core.dll, this assembly can cause crashes in the runtime.

Since the dll is unloaded, I cannot create a session with a server, since it causes a ReflectionTypeLoadException.

fejesgergorichard, 2020-09-21:
I managed to import the library via NuGet for Unity, but when creating a session I get the following Exception:
System.TypeInitializationException: The type initializer for ´Opc.Ua.EncodeableFactory' threw an exception.---> **System.MissingMethodException**: Default constructor not found

I have tried adding a **link.xml** as advised [here](https://github.com/OPCFoundation/UA-.NETStandard/issues/760#issuecomment-514502763) but then I cannot build my project. 

_(I also got about 10-12 error messages in Unity editor, when I hit play **in the editor, my script runs fine** and I can create/delete a session, read node values etc.)_ 

Any help, thought, opinion is appreciated.

Senethys, 2020-09-30:
Hi. I'm currently trying do to the same as you, but using assemblies from Traeger.de 
Have you had any success on this?

fejesgergorichard, 2020-10-13:
@Senethys  Hi, I decided to proceed on a different path and creating a UWP app for the OPC client.
It works on the HoloLens with absolutely no problem.

The "but THEN I cannot build" problem was about trying to preserve some types that cannot be preserved. (at least this is what I remember).

I also tried the traeger.de assemblies and they worked fine from the unity editor. However, when I tried to build the project, it was missing the HttpUtility package.

closed 2021-04-06 by AlinMoldovean