OPCFoundation/UA-.NETStandard #1116: Reduce assemblies to deploy

freica, 2020-09-16:
If I build an application, for example your quickstart Application "**Empty Server**" using the target **.NET Framework 4.6.2** I get 194 binary files to deploy, most of it have the namespace **System.xxx** or **Microssoft.xxx** which are allready installed on the target machines by the os or an Framework-Setup.
Using the tool "Microsoft.Packaging.Tools.Trimming" I can reduce the files to 54.
How can I reduce this files to a minimum?

MD-V, 2020-09-17:
Similar to 

https://github.com/OPCFoundation/UA-.NETStandard/issues/600
https://github.com/OPCFoundation/UA-.NETStandard/issues/1051
and 
https://github.com/OPCFoundation/UA-.NETStandard/issues/981

mregen, 2020-10-11:
Please try the latest release: https://github.com/OPCFoundation/UA-.NETStandard/releases/tag/1.4.363.107

freica, 2020-10-12:
I tried the release today.

If I build again your quickstart Application "Empty Server" using the target .NET Framework 4.6.2 I get 155 binary files to deploy, most of it have the namespace System.xxx or Microssoft.xxx which are allready installed on the target machines by the os or an Framework-Setup.
Using the tool "Microsoft.Packaging.Tools.Trimming" I can reduce the files to 56.
So I do not see any change on this.

mregen, 2020-12-13:
release 364 has the https binding as an optional component...

freica, 2020-12-23:
I tried it today again with V1.4.365.2
Here I see a great progess, using the tool "Microsoft.Packaging.Tools.Trimming" I get only  13 additional assemlies for the "Empty Server".

_I don't want to be petty now, but there are still 6 system assemblies left, e.g. System.Memory.dll, System.ValueTuple.dll, ....
Aren't they part of the already installed .NET framework?_
[image]

**Merry Christmas!**

mregen, 2020-12-23:
Hi @freica, thanks for trying ... Merry Christmas!

mregen, 2021-01-18:
Hi @freica, I think the few remainings are dependencies from the new [Asn1](https://www.nuget.org/packages/System.Formats.Asn1/) .NET Standard library.

mregen, 2021-01-28:
one more, see #1255 --> for .NET Core

mregen, 2021-04-01:
see #1279 it removes the dependency on servicemodel. -- there is no more improvement coming.

closed 2021-06-17 by mregen