OPCFoundation/UA-.NETStandard #1119: NuGet Documentations missing?

KevinValmo, 2020-09-21:
Hello, i'm **new**.

I downloaded the nuget and i started a simple project to make tests.
I noticed that there aren't intellisense suggestions.
Here are some img.

I'm doing something wrong or docs are missing in the nuget?

Thank you!
 
[image]

[image]

AlinMoldovean, 2020-09-22:
similar with #795

KevinValmo, 2020-09-22:
> similar with #795

Ok nice,
but i think `<GenerateDocumentationFile>true</GenerateDocumentationFile>` should be done in these ua projects, at least to provide a minimal documentation, that can however be extended in custom libraries. 
Shouldn't it? Or do you want to wait till the bug is fixed by Microsoft without using this workaround?

**Thank you for support.**

mregen, 2020-10-11:
Please try the latest release --> https://github.com/OPCFoundation/UA-.NETStandard/releases/tag/1.4.363.107

KevinValmo, 2020-10-14:
> Please try the latest release --> https://github.com/OPCFoundation/UA-.NETStandard/releases/tag/1.4.363.107

Hello @mregen.
I updated to last release (1.4.363.107) and, in the short test i made, documentations requested seems to be shown.
However many test has to come to test the full library.
Thank you.

closed 2020-10-14 by KevinValmo