OPCFoundation/UA-.NETStandard #1120: Implement subscription mechanism

soaibsafi, 2020-09-21:
I want to subscribe server data. So I defined a monitoredItem list to capture the notification. 

```
var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };
var list = new List<MonitoredItem> {
        new MonitoredItem(subscription.DefaultItem) { DisplayName = "Speed", StartNodeId = "ns=2;i=4"},
    };
Debug.Log(list[0].StartNodeId);
list.ForEach(i => i.Notification += OnNotification);
subscription.AddItems(list);
list.ForEach(i =>  Debug.Log(i.Subscription.Id));
```

When ever it get new notification, it suppose to call the `OnNotification` deligate. But it my case the `OnNotification` is never be called. 
```
private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
{
        Debug.Log("OnNotification");
        foreach (var value in item.DequeueValues())
        {
            Debug.LogFormat("{0}: {1}, {2}, {3}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
        }
}
```

How can I implement subscription mechanism in this case?

AlinMoldovean, 2020-09-21:
Hi @soaibsafi ,

The "`subscription.Create();`" call seems to be missing from you code.

aravinds4uAll, 2020-09-21:
Also please set subscription.PublishingEnabled =true.

soaibsafi, 2020-09-21:
Hello @AlinMoldovean & @aravinds4uAll 
Thank you for the suggestions. I have already added that. But it doesn't make cane change. 

```
var list = new List<MonitoredItem> {
        new MonitoredItem(subscription.DefaultItem) { DisplayName = "Speed", StartNodeId = "ns=2;i=4"},
    };
Debug.Log(list[0].StartNodeId);
list.ForEach(i => i.Notification += OnNotification);
subscription.AddItems(list);
list.ForEach(i =>  Debug.Log(i.Subscription.Id));

Debug.Log("Add the subscription to the session.");
subscription.PublishingEnabled = true;
session.AddSubscription(subscription);
subscription.Create();
```

Is there any problem with the **namespace** or **NodeId**? The `ns `and `i`is valid btw.

aravinds4uAll, 2020-09-21:
See if this Speed node value is a constantly changing one. Unless the value is changing, you won't get a notification. Go and change the value of node from OPC server and see whether you're getting notification after running your code. If that is also not working try to add servertime as a node that changes value every second. You should get the notification for it. If that doesn't work, please let me know.

soaibsafi, 2020-09-25:
Hello @aravinds4uAll 

My speed node value is changing in every second. Here is my test server.

```
import sys
sys.path.insert(0, "..")
import time

from opcua import ua, Server

if __name__ == "__main__":
    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://127.0.0.1:4840/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    # populating our address space
    myobj = objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "Speed", 150)
    myvar.set_writable()    # Set MyVariable to be writable by clients

    # starting!
    server.start()
    
    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            myvar.set_value(count)
            print(myvar.get_value())
    finally:
        #close connection, remove subcsriptions, etc
        server.stop()
```

And here is the debug output.
[image]

Is there any problem because of the Unity? I implemented the  substription inside the Unity `start()` function.

soaibsafi, 2020-09-25:
I found this from the OPC forum: 

- **The event on the subscription object has not been subscribe to.**

But I couldn't understand where to subscribe.

AlinMoldovean, 2020-10-01:
Hi @soaibsafi ,

I put an example for creating a Subscription and MonitoredItems from Client side.
Please check it here:
https://github.com/OPCFoundation/UA-.NETStandard/blob/54f7f8aa9b2e07531f73dadd6e4bd933566302ee/Applications/ConsoleReferenceClient/UAClient.cs#L387

soaibsafi, 2020-10-05:
Hello @AlinMoldovean and @aravinds4uAll 

I figured out my problem. My solution couldn't load the Configuration file. Hence, load the `LoadApplicationConfiguration()` method. But I am couldn't find out why this `LoadApplicationConfiguration()` is not working. 

**Here is my code-**
```
Debug.Log("1 - Create an Application Configuration.");
        exitCode = ExitCode.ErrorCreateApplication;

        ApplicationInstance application = new ApplicationInstance
        {
            ApplicationName = "UA Core Sample Client",
            ApplicationType = ApplicationType.Client,
            ConfigSectionName = Utils.IsRunningOnMono() ? "Opc.Ua.MonoSampleClient" : "Opc.Ua.SampleClient"
        };
        
        // load the application configuration.
        ApplicationConfiguration config = await application.LoadApplicationConfiguration("D:/InterfaceApp/InterfaceApp - 3/Assets/Opc.Ua.MonoSampleClient.Config.xml", false);
        Debug.Log(application.ConfigSectionName);
```
**And here is the output-**
I tried with both `Opc.Ua.MonoSampleClient.Config.xm`l and `Opc.Ua.SampleClient.Config.xml`. But the config file wasn't being load.

[image]

AlinMoldovean, 2020-10-06:
@soaibsafi ,

Could you try with [ConsoleReferenceClient](https://github.com/OPCFoundation/UA-.NETStandard/tree/54f7f8aa9b2e07531f73dadd6e4bd933566302ee/Applications/ConsoleReferenceClient) ?
It has a complete example for your requested workflow.

soaibsafi, 2020-10-09:
> @soaibsafi ,
> 
> Could you try with [ConsoleReferenceClient](https://github.com/OPCFoundation/UA-.NETStandard/tree/54f7f8aa9b2e07531f73dadd6e4bd933566302ee/Applications/ConsoleReferenceClient) ?
> It has a complete example for your requested workflow.

@AlinMoldovean ,
Yeah I tried the `ConsoleReferenceClient`.
This is working perfectly in Console Application. But whenever I tried to do the same thing in Unity, I got an `ServiceResultException:Could not load configuration file.`

I kept the `ConsoleReferenceClient.Config.xml` on the same directory where the `Program.cs` script exists

```
ServiceResultException: Could not load configuration file.
Opc.Ua.Configuration.ApplicationInstance+<LoadApplicationConfiguration>d__33.MoveNext () (at <1c5dac388c934c65a66a1f12b206c16d>:0)
Rethrow as AggregateException: One or more errors occurred.
System.Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) (at <fb001e01371b4adca20013e0ac763896>:0)
System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) (at <fb001e01371b4adca20013e0ac763896>:0)
System.Threading.Tasks.Task.Wait () (at <fb001e01371b4adca20013e0ac763896>:0)
TestClient.Start () (at Assets/Scripts/TestClient.cs:22)
```

AlinMoldovean, 2020-10-16:
moved to #1153

closed 2020-10-16 by AlinMoldovean