OPCFoundation/UA-.NETStandard #1133: OPC connection on how to slove Timeout problem??

shixiaoyan-1, 2020-10-06:
**when i decide to connect to OPC service Timeout Exception appears.

the details bellow:   I hope someone can help me .**

_EXCEPTION (ArgumentException)
--
Timeout must be greater than zero.参数名: timeout
STACK TRACE
在 Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest  request, Int32 timeout, AsyncCallback callback, Object state)在  Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.BeginSendRequest(IServiceRequest  request, AsyncCallback callback, Object callbackData)在  Opc.Ua.Bindings.UaSCUaBinaryTransportChannel.SendRequest(IServiceRequest  request)在 Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader,  String endpointUrl, StringCollection localeIds, StringCollection profileUris,  EndpointDescriptionCollection& endpoints)在  Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris)在  Opc.Ua.Client.CoreClientUtils.SelectEndpoint(String discoveryUrl, Boolean  useSecurity, Int32 discoverTimeout)在  OpcUaHelper.OpcUaClient.<Connect>d__2.MoveNext()--- 引发异常的上一位置中堆栈跟踪的末尾  ---在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task  task)在  System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task  task)在 System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()在  OpcUaHelper.OpcUaClient.<ConnectServer>d__1.MoveNext()---  引发异常的上一位置中堆栈跟踪的末尾 ---在  System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)在  System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task  task)在 System.Runtime.CompilerServices.TaskAwaiter.GetResult()在  caogao.Form1.<Form1_Load>d__3.MoveNext() 位置  E:\Projects\VS2017Document\WindowsFormsProjects\caogao\caogao\Form1.cs:行号  53_

mregen, 2020-10-07:
@shixiaoyan-1 could you check why this parameter is 0: Int32 timeout?

shixiaoyan-1, 2020-10-07:
> @shixiaoyan-1 could you check why this parameter is 0: Int32 timeout?

Now, i am not clearly about why it is int32, but everything is fine until MySQL connected . just means, these is no problem when i use opcuahelper only, but when i import MySql.Data   and updata some NuGet ,the program appears error above.
i do not know why? maybe the core2.0?

AlinMoldovean, 2020-10-08:
@shixiaoyan-1 ,

The error is caused because you are calling `CoreClientUtils.SelectEndpoint(String discoveryUrl, Boolean  useSecurity, Int32 discoverTimeout)` with value lower than 0 for `discoverTimeout `parameter.

The validation takes place here:
https://github.com/OPCFoundation/UA-.NETStandard/blob/c2c2801d3e42e19d19228a0fc592bba5766ecffe/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryClientChannel.cs#L270-L273

closed 2020-10-14 by mregen