OPCFoundation/UA-.NETStandard #1134: Endpoints exposed by the server configured with localhost may be not conformed with the OPCUA specs

GregoireG-C, 2020-10-07:
Hello there,

Extract from the spec (part 4: 5.4.2: FindServers):

> A server may have multiple HostNames. For this reason, the client shall pass the URL it used to connect to the Endpoint to this Service. The implementation of this Service. **The implementation of this Service shall use this information to return responses that are accessible to the Client via the provided URL.**

The stack behaviour:

In the case we configure a DNS hostname to "localhost" for the server and a client is trying to connect itself from a distant machine via the DNS containing the machine name of the server

The client will be able to coonect to the discovery server without any issue. Then the server will send the endpoints url containing DNS composed of "localhost".

Then the client will receive it and will change the DNS by the server machine name. (cf code below from Stack/Client/DiscoveryClient.cs)

```
public virtual EndpointDescriptionCollection GetEndpoints(StringCollection profileUris)
        {
            EndpointDescriptionCollection endpoints = null;

            GetEndpoints(
                null,
                this.Endpoint.EndpointUrl,
                null,
                profileUris,
                out endpoints);

            // if a server is behind a firewall, can only be accessed with a FQDN or IP address
            // it may return URLs that are not accessible to the client. This problem can be avoided 
            // by assuming that the domain in the URL used to call GetEndpoints can be used to 
            // access any of the endpoints. This code patches the returned endpoints accordingly.
            Uri endpointUrl = Utils.ParseUri(this.Endpoint.EndpointUrl);
            if (endpointUrl != null)
            {
                // patch discovery Url to endpoint Url used for service call
                foreach (EndpointDescription discoveryEndPoint in endpoints)
                {
                    Uri discoveryEndPointUri = Utils.ParseUri(discoveryEndPoint.EndpointUrl);
                    if (endpointUrl.Scheme == discoveryEndPointUri.Scheme)
                    {
                        UriBuilder builder = new UriBuilder(discoveryEndPointUri);
                        builder.Host = endpointUrl.DnsSafeHost;
                        builder.Port = endpointUrl.Port;
                        discoveryEndPoint.EndpointUrl = builder.ToString();
                    }

                    if (discoveryEndPoint.Server != null &&
                        discoveryEndPoint.Server.DiscoveryUrls != null)
                    {
                        discoveryEndPoint.Server.DiscoveryUrls.Clear();
                        discoveryEndPoint.Server.DiscoveryUrls.Add(this.Endpoint.EndpointUrl.ToString());
                    }
                }
            }

            return endpoints;
        }
```

This works **BUT** I don't think this behaviour follows the OPCUA specifications since the server provides a URL that the client must transform to be able to connect.

I understand why you choose it since the server provided all the information necessary for the client to connect. Moreover, this kind of behaviour would avoid DDOS. So it makes more sense than the OPCUA specifications.

Then I think you should change the specification to make it clear that it's the client task to adjust the url provided by the discovery server.
Otherwise, the discovery server should send url endpoints with its machine name as DNS.

I hope i didn't missed anything from the specs.

Thanks for your help :)

GregoireG-C, 2020-10-07:
To follow the OPCUA specifications.

In Stack/Server/ServerBase.cs we should have:

```
/// <summary>
/// Initializes the list of base addresses.
/// </summary>
private void InitializeBaseAddresses(ApplicationConfiguration configuration)
{
            BaseAddresses = new List<BaseAddress>();

            StringCollection sourceBaseAddresses = null;
            StringCollection sourceAlternateAddresses = null;

            if (configuration.ServerConfiguration != null)
            {
                sourceBaseAddresses = configuration.ServerConfiguration.BaseAddresses;
                sourceAlternateAddresses = configuration.ServerConfiguration.AlternateBaseAddresses;
            }

            if (configuration.DiscoveryServerConfiguration != null)
            {
                sourceBaseAddresses = configuration.DiscoveryServerConfiguration.BaseAddresses;
                sourceAlternateAddresses = configuration.DiscoveryServerConfiguration.AlternateBaseAddresses;
            }

            if (sourceBaseAddresses == null)
            {
                return;
            }

            foreach (string baseAddress in sourceBaseAddresses)
            {
                BaseAddress address = new BaseAddress() { Url = new Uri(baseAddress) };

                //MY CODE
                //If the address contains localhost, we can change it with the dnsMachine
                String dnsMachine = NormalizeHostname("localhost").Result;
                if (NormalizeHostname(address.Url.DnsSafeHost).Result == dnsMachine)
                {
                    address = new BaseAddress() { Url = new Uri(baseAddress.Replace(address.Url.DnsSafeHost, dnsMachine)) };
                }
                //END

                if (sourceAlternateAddresses != null)
                {
                    foreach (string alternateAddress in sourceAlternateAddresses)
                    {
                        Uri alternateUrl = new Uri(alternateAddress);

                        if (alternateUrl.Scheme == address.Url.Scheme)
                        {
                            if (address.AlternateUrls == null)
                            {
                                address.AlternateUrls = new List<Uri>();
                            }

                            address.AlternateUrls.Add(alternateUrl);
                        }
                    }
                }

                switch (address.Url.Scheme)
                {
                    case Utils.UriSchemeHttps:
                    {
                        address.ProfileUri = Profiles.HttpsBinaryTransport;
                        address.DiscoveryUrl = address.Url;
                        break;
                    }

                    case Utils.UriSchemeOpcTcp:
                    {
                        address.ProfileUri = Profiles.UaTcpTransport;
                        address.DiscoveryUrl = address.Url;
                        break;
                    }
                }

                BaseAddresses.Add(address);
            }
}
```

mrsuciu, 2021-09-10:
The existing solution is a workaround that probably is not 100% compliant but improves the interoperability with many of the faulty servers.

mrsuciu, 2021-09-10:
Please reopen and provide more information if additional issues are found.

closed 2021-09-10 by mrsuciu