OPCFoundation/UA-.NETStandard #1139: Problem with Reference Client (UA.-.NETStandard-master)

BeckhoffPLC, 2020-10-08:
Hello everybody, I downloaded the OPC UA standard project from github (https://github.com/OPCFoundation/UA-.NETStandard).
I would like to use the Reference Client there to access a Beckhoff PLC (server).
To do this, I opened the project in VS2017 and set the Reference Client as the startup project. Then I started the project with F5.
The following window opened:
[image]
If I now specify the IP address and the port of my PLC (as shown above) and then click on "Connect",
the client sometimes establishes a connection to the PLC and sometimes I get the following error message:
[image]
In about 80% of the cases there is this error message and the client does not connect to the server.
In about 20% of the cases a connection to the server is established and I can read and write variables from the server.
I didn't change the program and it seems completely arbitrary when the connection can be established and when not.
Has anyone ever had a similar problem?

Many thanks in advance.

AlinMoldovean, 2020-11-12:
Hi @BeckhoffPLC ,

I checked this and the reported issue does not replicate when testing with several OPC UA servers including ReferenceServer from this repo.
The _BadNotConnected_ error during client session connect occurs in case the server endpoint is not available or the server refuses the connection.
This should be investigates on the server side. Please check it with the UA Server provider, i.e. Beckhoff PLC UA Server.

BeckhoffPLC, 2020-11-12:
Thanks for the answer. I also posted the question in the Stackoverflow forum ([https://stackoverflow.com/questions/64313719/reference-client-ua-netstandard-master-is-badnotconnected](url)) because nobody has answered me here yet. In the forum I have already described some things that I have tried. The Beckhoff PLC was not the cause of the problem. I found the cause and described it in Stackoverflow, but I couldn't solve the problem.

AlinMoldovean, 2021-03-16:
@BeckhoffPLC , please reopen if still not solved.

closed 2021-03-16 by AlinMoldovean