OPCFoundation/UA-.NETStandard #1144: How to add a node using the client?

davidguidali, 2020-10-09:
Is it possible to add a static node and set the value from within the client?
I did not find any examples for this. There is the session.AddNodes method, but I guess this is not the one I need?
Or do I need to create some kind of API on my server to be able to generate nodes from outside of it?
Please provide me information or point me to the docs I missed.

Thanks!

AlinMoldovean, 2020-10-12:
Hi @davidguidali ,

You need to implement the AddNodes service on server side for this.
It requires an override of `AddNodes `in _public partial class StandardServer : SessionServerBase_

https://github.com/OPCFoundation/UA-.NETStandard/blob/7fe666e9bb4a4cbb6e3ccfcec432c9874f85c9f3/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.ServerBase.cs#L676-L690

davidguidali, 2020-10-12:
Hi @AlinMoldovean 
Thank you very much for your answer!
Do you suggest to create a new node manager derived from CustomNodeManager2, implement an AddNodes() service there and call this from my CustomServer? 
I mean the logic should be implemented in the node manager, not in the server, am I right on this?

AlinMoldovean, 2020-10-12:
Hi @davidguidali,

Yes, first you need to implement the handling of the service call in the custom Server, and then pass the call to a NodeManager that implements the logic for `AddNodes()`.
Some code sample:

For CustomServer:
```
        NodeManagementNodeManager m_nodeManagementNodeManager;

        public override ResponseHeader AddNodes(
            RequestHeader requestHeader,
            AddNodesItemCollection nodesToAdd,
            out AddNodesResultCollection results,
            out DiagnosticInfoCollection diagnosticInfos)
        {
            OperationContext context = ValidateRequest(requestHeader, RequestType.AddNodes);

            try
            {
                if (nodesToAdd == null || nodesToAdd.Count == 0)
                {
                    throw new ServiceResultException(StatusCodes.BadNothingToDo);
                }

                m_nodeManagementNodeManager.AddNodes(
                    context,
                    nodesToAdd,
                    out results,
                    out diagnosticInfos);

                return CreateResponse(requestHeader, context.StringTable);
            }
            catch (ServiceResultException e)
            {
                lock (ServerInternal.DiagnosticsLock)
                {
                    ServerInternal.ServerDiagnostics.RejectedRequestsCount++;

                    if (IsSecurityError(e.StatusCode))
                    {
                        ServerInternal.ServerDiagnostics.SecurityRejectedRequestsCount++;
                    }
                }

                throw TranslateException(context, e);
            }
            finally
            {
                OnRequestComplete(context);
            }
        }
```

and for NodeManager:

```
        public void AddNodes(OperationContext context, AddNodesItemCollection nodesToAdd, out AddNodesResultCollection results, out DiagnosticInfoCollection diagnosticInfos)
        {
            // Validate nodesToAdd parameter
            if (nodesToAdd == null)
            {
                throw new ServiceResultException(StatusCodes.BadInvalidArgument, "The nodesToAdd parameter is null.");
            }

            // Create result lists
            results = new AddNodesResultCollection(nodesToAdd.Count);
            diagnosticInfos = new DiagnosticInfoCollection(nodesToAdd.Count);

            for (int ii = 0; ii < nodesToAdd.Count; ii++)
            {
                // Call AddNode and update results
                AddNodesResult addResult;
                DiagnosticInfo diagnosticInfo;

                AddNode(context, nodesToAdd[ii], out addResult, out diagnosticInfo);

                results.Add(addResult);
                diagnosticInfos.Add(diagnosticInfo);
            }
        }

        private void AddNode(OperationContext context, AddNodesItem nodeToAdd, out AddNodesResult result, out DiagnosticInfo diagnosticInfo)
        {
            result = new AddNodesResult();
            diagnosticInfo = new DiagnosticInfo();

            try
            {
                // Pre-validate the request
                ServiceResult error = ValidateAddNodesItem(context, nodeToAdd);

                if (ServiceResult.IsBad(error))
                {
                    result.StatusCode = error.Code;

                    // Add diagnostics if requested
                    if ((context.DiagnosticsMask & DiagnosticsMasks.OperationAll) != 0)
                    {
                        diagnosticInfo = new DiagnosticInfo(error, context.DiagnosticsMask, false, new StringTable());
                    }

                    return;
                }

         // TODO: Add node in Address space

       }
```

davidguidali, 2020-10-12:
@AlinMoldovean Thanks very much, exactly what I needed! :)
Gonna try it out

davidguidali, 2020-10-22:
Perfect, I did it quite similar and it worked like a charm. Thanks.

closed 2020-10-22 by davidguidali

YasminAkaichi, 2021-06-21:
I had an exception that says that m_nodeManagementNodeManager is null and this field is never assigned to. Can you tell me please what should I do about it ? Thanks

YasminAkaichi, 2021-06-21:
@AlinMoldovean

AlinMoldovean, 2021-06-22:
@YasminAkaichi ,

The code I placed here is just a sample code.
You need to further implement "**a NodeManager that implements the logic for AddNodes().**" and set it as _**m_nodeManagementNodeManager**_

YasminAkaichi, 2021-06-29:
@AlinMoldovean  Thank you but it didnt work for me.

PopeyeZhong, 2025-04-20:
@AlinMoldovean 

Hi AlinMoldovean:

First of all, thank you very much for the sample code.

I need to dynamically create a Folder node while Server is running. Below is the section I added based on your code:

```csharp
class MyNodeManager : CustomNodeManager2
{
	private (AddNodesResult result, DiagnosticInfo diagnostic) AddNode(OperationContext context, AddNodesItem node)
	{
		if (node.NodeClass == NodeClass.Object)
		{
			if (node.TypeDefinition == ObjectTypeIds.FolderType)
			{
				var displayName = node.BrowseName.Name;
				var description = (string)null;

				if (node.NodeAttributes?.Body is ObjectAttributes attributes)
				{
					displayName = attributes.DisplayName?.Text;
					description = attributes.Description?.Text;
				}

				var root = this.Server.CoreNodeManager.Find(ObjectIds.ObjectsFolder, null);
				var folder = this.CreateFolder(null, node.RequestedNewNodeId, node.BrowseName.Name, displayName, description);
				folder.AddReference(ReferenceTypes.Organizes, true, ObjectIds.ObjectsFolder);
				this.AddPredefinedNode(this.SystemContext, folder);

				return (new AddNodesResult()
				{
					StatusCode = StatusCodes.Good,
					AddedNodeId = folder.NodeId,
				}, null);
			}
			else
			{
				throw new ServiceResultException(StatusCodes.BadTypeDefinitionInvalid);
			}
		}

		throw new ServiceResultException(StatusCodes.BadNodeClassInvalid);
	}

	private FolderState CreateFolder(FolderState parent, ExpandedNodeId id, string name, string displayName, string description)
	{
		var nodeId = id == null || id.IsNull ? new NodeId(Guid.NewGuid(), this.NamespaceIndex) : (NodeId)id;

		var folder = new FolderState(parent)
		{
			SymbolicName = name,
			ReferenceTypeId = ReferenceTypes.Organizes,
			TypeDefinitionId = ObjectTypeIds.FolderType,
			NodeId = nodeId,
			BrowseName = new QualifiedName(name, this.NamespaceIndex),
			DisplayName = displayName,
			Description = description,
			WriteMask = AttributeWriteMask.None,
			UserWriteMask = AttributeWriteMask.None,
			EventNotifier = EventNotifiers.None
		};

		if (parent != null)
			parent.AddChild(folder);

		return folder;
	}
}
```

The problem with it is that it runs fine, but the added Folder node is not visible in the Unified Automation UaExpert. If you rewrite the `NodeStateCollection LoadPredefinedNodes(ISystemContext context)` method, it is possible to observe that the Folder node is added successfully with the same code as above.

I guess it's because the default `AddPredefinedNode(ISystemContext context, NodeState node)` method doesn't support adding Folder nodes dynamically on the server running?

May I know how to implement this function correctly? thanks...